robust-rosin/robust #144: Error reproducing Kobuki 45ee84a

git-afsantos, 2018-10-15:
The issue is very straightforward. It is a copy-and-paste mistake in the CMake of `kobuki_ftdi`.

```cmake
catkin_package(
   INCLUDE_DIRS include
   LIBRARIES kobuki_ros kobuki_nodelet
   CATKIN_DEPENDS ecl_command_line
   DEPENDS libusb libftdi
)
```

The `LIBRARIES` line was unintentionally left there, when it should not be there at all.

From what I understand, this will manifest as a build error, and I was able to build the Docker container without issues. When running the container, I am getting a different build error:

> [ 10%] Building CXX object repo-under-test/kobuki_ftdi/src/CMakeFiles/find_devices.dir/find_devices.cpp.o
> In file included from /ros_ws/src/repo-under-test/kobuki_ftdi/src/find_devices.cpp:16:0:
> /ros_ws/src/repo-under-test/kobuki_ftdi/include/kobuki_ftdi/scanner.hpp:45:17: fatal error: usb.h: No such file or directory
> `#include <usb.h>`
> compilation terminated.
> make[2]: *** [repo-under-test/kobuki_ftdi/src/CMakeFiles/find_devices.dir/find_devices.cpp.o] Error 1
> make[1]: *** [repo-under-test/kobuki_ftdi/src/CMakeFiles/find_devices.dir/all] Error 2
> make: *** [all] Error 2
> Invoking "make -j1 -l1" failed

This happens when I run

```bash
./unfix
./build.sh
```

*and* when I run

```bash
./fix
./build.sh
```

I am guessing it is probably the `.rosinstall` that is missing something here.

@gavanderhoorn @ChrisTimperley any ideas?

gavanderhoorn, 2018-10-15:
If you can show the output of the Docker build we could see whether `libusb-dev` is installed or not.

It's certainly a `build_depend` of `kobuki_ftdi` ([here](https://github.com/yujinrobot/kobuki_core/blob/45ee84aa1e920b26f8191763e20c10c5806ead60/kobuki_ftdi/package.xml#L21)).

gavanderhoorn, 2018-10-15:
`tee` can be very helpful in these cases.

gavanderhoorn, 2018-10-15:
> I am guessing it is probably the `.rosinstall` that is missing something here.

btw: `libusb` is a system dependency, not a ROS pkg, so it will never end up in the `.rosinstall` file.

It should be installed by `rosdep` for you.

git-afsantos, 2018-10-15:
Just did it again, and a search for `usb` only finds the catkin output.

> -- +++ processing catkin package: 'kobuki_ftdi'
> -- ==> add_subdirectory(repo-under-test/kobuki_ftdi)
> -- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
> -- checking for one of the modules 'libusb'
> -- checking for one of the modules 'libftdi'
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /ros_ws/build
> Scanning dependencies of target find_devices
> [ 10%] Building CXX object repo-under-test/kobuki_ftdi/src/CMakeFiles/find_devices.dir/find_devices.cpp.o
> In file included from /ros_ws/src/repo-under-test/kobuki_ftdi/src/find_devices.cpp:16:0:
> /ros_ws/src/repo-under-test/kobuki_ftdi/include/kobuki_ftdi/scanner.hpp:45:17: fatal error: usb.h: No such file or directory
>  `#include <usb.h>`

What else should I provide you, or be looking for?

gavanderhoorn, 2018-10-15:
Can you show me the section of the Docker build output where it runs `rosdep install ..`?

git-afsantos, 2018-10-15:
> Step 30/41 : RUN apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*
> ---> Running in e6a95c3a0b0f
>Get:1 http://security.ubuntu.com trusty-security InRelease [65.9 kB]
Ign http://archive.ubuntu.com trusty InRelease
Get:2 http://packages.osrfoundation.org trusty InRelease [4255 B]
Get:3 http://archive.ubuntu.com trusty-updates InRelease [65.9 kB]
Get:4 http://packages.osrfoundation.org trusty/main amd64 Packages [38.2 kB]
Get:5 http://security.ubuntu.com trusty-security/universe Sources [103 kB]
Get:6 http://archive.ubuntu.com trusty-backports InRelease [65.9 kB]
Get:7 http://security.ubuntu.com trusty-security/main amd64 Packages [967 kB]
Get:8 http://archive.ubuntu.com trusty Release.gpg [933 B]
Get:9 http://archive.ubuntu.com trusty-updates/universe Sources [267 kB]
Get:10 http://archive.ubuntu.com trusty-updates/main amd64 Packages [1387 kB]
Get:11 http://security.ubuntu.com trusty-security/restricted amd64 Packages [18.1 kB]
Get:12 http://security.ubuntu.com trusty-security/universe amd64 Packages [335 kB]
Get:13 http://security.ubuntu.com trusty-security/multiverse amd64 Packages [4722 B]
Get:14 http://archive.ubuntu.com trusty-updates/restricted amd64 Packages [21.4 kB]
Get:15 http://archive.ubuntu.com trusty-updates/universe amd64 Packages [631 kB]
Get:16 http://archive.ubuntu.com trusty-updates/multiverse amd64 Packages [16.0 kB]
Get:17 http://archive.ubuntu.com trusty-backports/main amd64 Packages [14.7 kB]
Get:18 http://archive.ubuntu.com trusty-backports/restricted amd64 Packages [40 B]
Get:19 http://archive.ubuntu.com trusty-backports/universe amd64 Packages [52.5 kB]
Get:20 http://archive.ubuntu.com trusty-backports/multiverse amd64 Packages [1392 B]
Get:21 http://archive.ubuntu.com trusty Release [58.5 kB]
Get:22 http://archive.ubuntu.com trusty/universe Sources [7926 kB]
Get:23 http://archive.ubuntu.com trusty/main amd64 Packages [1743 kB]
Get:24 http://archive.ubuntu.com trusty/restricted amd64 Packages [16.0 kB]
Get:25 http://archive.ubuntu.com trusty/universe amd64 Packages [7589 kB]
Get:26 http://archive.ubuntu.com trusty/multiverse amd64 Packages [169 kB]
Fetched 21.6 MB in 42s (509 kB/s)
Reading package lists...
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
> Recommended: please run
>
>	`rosdep update`
>
> [91mWarning: running 'rosdep update' as root is not recommended.[0m[91m
[0m[91m  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.[0m[91m
[0mreading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
Add distro "melodic"
updated cache in /root/.ros/rosdep/sources.cache
Reading package lists...
Building dependency tree...
Reading state information...
Suggested packages:
  python-mock-doc
The following NEW packages will be installed:
  python-mock
0 upgraded, 1 newly installed, 0 to remove and 13 not upgraded.
Need to get 23.8 kB of archives.
After this operation, 140 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu/ trusty/main python-mock all 1.0.1-3 [23.8 kB]
Fetched 23.8 kB in 0s (123 kB/s)
Selecting previously unselected package python-mock.
(Reading database ... 17644 files and directories currently installed.)
Preparing to unpack .../python-mock_1.0.1-3_all.deb ...
Unpacking python-mock (1.0.1-3) ...
Setting up python-mock (1.0.1-3) ...
[... 27 lines cut ...]
Preparing to unpack .../libgtest-dev_1.6.0-1ubuntu6_all.deb ...
Unpacking libgtest-dev (1.6.0-1ubuntu6) ...
Setting up libgtest-dev (1.6.0-1ubuntu6) ...
Reading package lists...
Building dependency tree...
Reading state information...
Suggested packages:
  python-coverage python-nose-doc
The following NEW packages will be installed:
  python-nose
0 upgraded, 1 newly installed, 0 to remove and 13 not upgraded.
Need to get 119 kB of archives.
After this operation, 592 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu/ trusty/main python-nose all 1.3.1-2 [119 kB]
Fetched 119 kB in 0s (395 kB/s)
Selecting previously unselected package python-nose.
(Reading database ... 17731 files and directories currently installed.)
Preparing to unpack .../python-nose_1.3.1-2_all.deb ...
Unpacking python-nose (1.3.1-2) ...
Setting up python-nose (1.3.1-2) ...
[1mexecuting command [apt-get install -y python-mock][0m
[1mexecuting command [apt-get install -y python-empy][0m
[1mexecuting command [apt-get install -y libgtest-dev][0m
[1mexecuting command [apt-get install -y python-nose][0m
#All required rosdeps installed successfully

gavanderhoorn, 2018-10-15:
Hm. I believe this is missing the `stderr` output. `rosdep` prints to `stderr`, so we're missing the commands here.

Can you re-run with something like `....  2>&1 | tee build.log`?

git-afsantos, 2018-10-15:
The output is exactly the same, at least for that build step.

gavanderhoorn, 2018-10-15:
Hm.

That is strange.

What is the output of `dpkg -l | grep usb` in the container? Nothing probably?

git-afsantos, 2018-10-15:
There is output, actually.

```bash
root@8741d78734d9:/ros_ws# dpkg -l | grep usb
ii  libusb-0.1-4:amd64              2:0.1.12-23.3ubuntu1                       amd64        userspace USB programming library
```

gavanderhoorn, 2018-10-16:
That is only the runtime lib, not the dev headers.

I'm not sure what is going on here. I can try to reproduce.