robust-rosin/robust #145: Docker fails to build for Kobuki 9c8abeb

git-afsantos, 2018-10-15:
Docker command:

```bash
docker build -t robustrosin/test-9c8abeb \
-f ../../Dockerfile \
--build-arg UBUNTU_VERSION=precise \
--build-arg IS_BUILD_FAILURE=yes \
--build-arg USE_APT_OLD_RELEASES=no \
--build-arg ROS_DISTRO=hydro \
--build-arg CATKIN_PKG=kobuki_gazebo_plugins \
--build-arg REPO_FORK_URL=https://github.com/robust-rosin/kobuki_desktop \
--build-arg REPO_BUG_COMMIT=024d79e2204a2be7ac4284bea5da2ac6ff92262a \
--build-arg REPO_FIX_COMMIT=71f3294694ffda0e2ab48ba0842f71913204ac60 \
. 2>&1 | tee build.log
```

Error message:

> E: Unable to locate package libpcl-1.7-all-dev
E: Couldn't find any package by regex 'libpcl-1.7-all-dev'
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y libpcl-1.7-all-dev] failed
executing command [apt-get install -y libeigen3-dev]
executing command [apt-get install -y libapr1-dev]
executing command [apt-get install -y libaprutil1-dev]
executing command [apt-get install -y graphviz]
executing command [apt-get install -y libjasper-dev]
executing command [apt-get install -y python-support]
executing command [apt-get install -y python-imaging]
executing command [apt-get install -y libpcl-1.7-all-dev]
The command '`/bin/sh -c apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*`' returned a non-zero code: 1

@ChrisTimperley @gavanderhoorn , any ideas why PCL might be missing?
[ROS Index](http://rosindex.github.io/d/libpcl-all-dev/) says it should be there for Ubuntu Precise.

gavanderhoorn, 2018-10-16:
I believe that package is only provided by the OSRF buildfarm, not upstream Ubuntu.

gavanderhoorn, 2018-10-16:
And the index you linked @git-afsantos has not been updated for quite some time.

The new link is: https://index.ros.org

git-afsantos, 2018-10-16:
Do I have to manually add the pcl package to `.rosdistro`, then?

gavanderhoorn, 2018-10-16:
No.

We'd probably need to add the OSRF repositories to the Docker file.

We already add the Gazebo repositories ([here](https://github.com/robust-rosin/robust/blob/149edf7ac81bea990df5b83c036a34bc00873a69/Dockerfile#L105-L107)), so we should add `packages.ros.org`, but I don't want to add it as it could interfere with our build-from-sources approach.

`libpcl-1.7-all-dev` is located [here](http://packages.ros.org/ros/ubuntu/pool/main/p/pcl-1.7).

gavanderhoorn, 2018-10-16:
The thing is that we tried to keep the OSRF repositories out so as to not import pkgs that we're building from sources or to install all sorts of rolling dependencies.

However, for this dependency I don't really see a way around it: that `.deb` pkg is only provided by that repository.

It's the same issue as with Gazebo: we don't want that repository enabled in images either, but we cannot feasibly build everything from sources there. Especially not if it's not the PUT itself, but a (far removed) dependency.

Perhaps @ChrisTimperley has some idea.

ChrisTimperley, 2018-10-16:
> The thing is that we tried to keep the OSRF repositories out so as to not import pkgs that we're building from sources or to install all sorts of rolling dependencies.
>
> However, for this dependency [...]

I guess that we could add another build argument and use it specify whether or not to include OSRF repos?

gavanderhoorn, 2018-10-16:
> I guess that we could add another build argument and use it specify whether or not to include OSRF repos?

Yes, we could do that.

Would be nice to make adding the Gazebo repositories depend on a similar build arg then.

I'm not too happy about it, but it would seem to be the quickest way to get this to build for @git-afsantos,