robust-rosin/robust #150: Add support for missing dependencies

ChrisTimperley, 2018-10-23:
See #126, #131, #127, #128. 

Possible approaches:

1. Each bug provides its own `deps.rosinstall.diff` patch file, which is applied to `deps.rosinstall` to obtain a complete list of dependencies. Has a number of cons: (1) tedious to determine appropriate versions, especially for recursive cases, (2) requires that we add a no-op file to every bug directory to ensure that bugs build, (3) more complicated (and needlessly so) than simply giving a list of additional dependencies.
2. Each bug provides its own `extra.rosinstall` file, which provides a `rosintall` file containing all of the missing dependencies. Slightly nicer than (1) but still tedious and introduces additional files.
3. Update `time-machine` section in bug description file to contain a list that details the missing dependencies for a particular bug:
```
missing-dependencies:
  - tar:
    local-name: actionlib
    uri: https://github.com/ros-gbp/actionlib-release/archive/release/hydro/actionlib/1.10.3-0.tar.gz
    version: actionlib-release-release-hydro-actionlib-1.10.3-0
```
  The contents of the `missing-dependencies` section can then either be automatically written to `extra.rosinstall` or folded into `deps.rosinstall`.
4. Update `time-machine` section in bug description file to state the names of the missing dependencies:
```
missing-dependencies:
  - actionlib
```
The time machine helper script, `build-time-machine.py`, should query the time machine to determine the most appropriate release of each missing dependency (along with any recursive dependencies). The complete set of dependencies should be folded into `deps.rosinstall`.

gavanderhoorn, 2018-10-23:
I would suggest we go with your suggestion `4`: the rewrite of the time machine actually supports specifying multiple pkgs that it should generate a `.rosinstall` file for.

That would seem to automatically take care of everything as `rosinstall_generator` will do the-right-thing (ie: take (recursive) dependencies into account).

gavanderhoorn, 2018-10-23:
> The time machine helper script, `build-time-machine.py`, should query the time machine to determine the most appropriate release of each missing dependency (along with any recursive dependencies). The complete set of dependencies should be folded into `deps.rosinstall`.

This should not be necessary btw.

If the additional pkgs are included in the `rosinstall_generator` call, everything should be taken care of.

commit 1344bb46 (2018-11-21): add support for missing deps (fixes #150)
commit f61ebc84 (2018-11-24): Updated build-rosinstall.py to time machine v2 (fixes #172, fixes #150) (#173)

* fixed dir_bug for new directory structure

* remove dir creation code

* added Docker image checking

* check for existence of binary

* added requirements.txt

* updated time machine command

* updated script execution

* write contents to deps.rosinstall

* added repository renaming fix

* removed unnecessary vars

* added handling of GitHub issues

* multiple ROS packages are now supported

* handle stdin, stderr and stdout

* add support for missing deps (fixes #150)

* described missing-dependencies

* prefe [... cut]
commit c01a6d3f (2019-03-06): Fixed handling of missing dependencies (fixes #150) (#254)

* fix whitespace

* tweaked handling of PUTs

* fixed handling of missing dependencies

* remove old contents line

* remove unnecessary import

* report warning and add comment to .rosinstall file

* Make geometry2 rename notice more descriptive
closed 2019-03-06 by ChrisTimperley