robust-rosin/robust #151: Docker build fails for Turtlebot 61a75df

git-afsantos, 2018-10-24:
> E: Unable to locate package libpyside2-dev
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y libpyside2-dev] failed
[...]
The command `/bin/sh -c apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*` returned a non-zero code: 1

I found https://github.com/ros-infrastructure/rosdep/issues/465 which says that the deb is found at packages.ros.org, so I modified the docker command to use the OSRF repos, but it still fails at this point.

@gavanderhoorn @ChrisTimperley

gavanderhoorn, 2018-10-24:
can you post the full build output (in a [details section](https://gist.github.com/ericclemmons/b146fe5da72ca1f706b2ef72a20ac39d))?

gavanderhoorn, 2018-10-24:
I don't see `packages.ros.org` in the output of `apt-get update`:

```
Step 32/43 : RUN apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*
 ---> Running in bbfc1f11c3ae
Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [107 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial InRelease [247 kB]
Get:3 http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial InRelease [4255 B]
Get:4 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]
Get:5 http://archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]
Get:6 http://security.ubuntu.com/ubuntu xenial-security/universe Sources [97.1 kB]
Get:7 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [732 kB]
Get:8 http://archive.ubuntu.com/ubuntu xenial/universe Sources [9802 kB]
Get:9 http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial/main amd64 Packages [22.3 kB]
Get:10 http://security.ubuntu.com/ubuntu xenial-security/restricted amd64 Packages [12.7 kB]
Get:11 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [497 kB]
Get:12 http://security.ubuntu.com/ubuntu xenial-security/multiverse amd64 Packages [3747 B]
Get:13 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages [1558 kB]
Get:14 http://archive.ubuntu.com/ubuntu xenial/restricted amd64 Packages [14.1 kB]
Get:15 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages [9827 kB]
Get:16 http://archive.ubuntu.com/ubuntu xenial/multiverse amd64 Packages [176 kB]
Get:17 http://archive.ubuntu.com/ubuntu xenial-updates/universe Sources [283 kB]
Get:18 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [1122 kB]
Get:19 http://archive.ubuntu.com/ubuntu xenial-updates/restricted amd64 Packages [13.1 kB]
Get:20 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [896 kB]
Get:21 http://archive.ubuntu.com/ubuntu xenial-updates/multiverse amd64 Packages [18.8 kB]
Get:22 http://archive.ubuntu.com/ubuntu xenial-backports/main amd64 Packages [7965 B]
Get:23 http://archive.ubuntu.com/ubuntu xenial-backports/universe amd64 Packages [8532 B]
Fetched 25.7 MB in 21s (1179 kB/s)
Reading package lists...
```

as that is the repository that provides `libpyside2-dev`, that would be why `rosdep` can't install it.

Is the fix @ChrisTimperley added to use `packages.ros.org` present in the version you are using?

gavanderhoorn, 2018-10-24:
Perhaps this is a Docker caching issue again:

```
Step 28/43 : RUN if [ "${USE_OSRF_REPOS}" = "True" ]; then          echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/ros-latest.list        && wget http://packages.ros.org/ros.key -O - | apt-key add -     ; fi
 ---> Using cache
```

ChrisTimperley, 2018-10-24:
Just to confirm, what arguments are you supplying to `docker build`, @git-afsantos?

git-afsantos, 2018-10-24:
I am sure I cleared the cache before, but I am doing it again, just to be sure.

@ChrisTimperley I was just looking into that. I made sure I am in the `osrf-arg` branch, and I was looking into the Dockerfile, to make sure it is the version that accepts the new build argument.

I just noticed that I was supplying a `yes` value (to be consistent with the other args), but the Dockerfile expects `True`. I am running it again with `True`.

ChrisTimperley, 2018-10-24:
> I am sure I cleared the cache before, but I am doing it again, just to be sure.
>
> @ChrisTimperley I was just looking into that. I made sure I am in the `osrf-arg` branch, and I was looking into the [...]

Interesting. I didn't notice that we were mixing `True` and `yes` in the Dockerfile.

git-afsantos, 2018-10-24:
Switching the argument to `True` fixed it, the build now finishes cleanly.

closed 2018-10-24 by git-afsantos