robust-rosin/robust #153: Integrate jsonschema checking into CI to validate bug description files

ChrisTimperley, 2018-10-24:


gavanderhoorn, 2018-10-26:
I've enabled travis on this org, so we can actually do this.

re: json checking: perhaps add [yamllint](http://manpages.ubuntu.com/manpages/xenial/man1/yamllint.1.html) as well? Has saved me from indentation mistakes countless times.

gavanderhoorn, 2018-10-26:
With #154 on the way: do we want to enforce checks for all pushes to the repository? Only real way we can do that is to restrict direct pushing to branches and requiring pull requests.

Then we can require Travis to complete successfully.

Would be nice to avoid issues with incorrect `.bug` markup.

gavanderhoorn, 2018-10-26:
So from the two :+1: I got I see at least @ChrisTimperley and @git-afsantos are ok with restricting changes only being accepted through PRs.

I noticed @wasowski using direct pushes a lot and configuring things to have to go through PRs might annoy him.

@wasowski: any opinion?

gavanderhoorn, 2018-11-26:
I'm going to disable direct pushes to `master`, seeing as I've got 2 :+1:s for this.

We're not really introducing heaps of bad commits (I can only remember 46c224a0), but it would be good practice to only allow contributions through PRs that have been CI checked in any case.