robust-rosin/robust #157: How should we deal with code smells and inefficiencies?

ChrisTimperley, 2018-10-31:
See the following for an example: https://github.com/robust-rosin/robust/blob/master/mavros/0e2ea0c.bug

git-afsantos, 2018-10-31:
I have come across a few similar bugs myself. I remember that, in one of the earlier meetings, we agreed to focus on `error` bugs first, and only later on try to reproduce smells, as we go down the list. I believe we did not set a methodology to reproduce them, though.

Testing this would be easier if it were a case of *you are using the bad/deprecated function X, use Y instead*. In your case, the problem is a *combination* of `size` and the `>` operator, and `size` could be legitimately used elsewhere.

I think this will be a similar case to the one I brought up some time ago, and the answer is that figuring out a test for this is more trouble than it is worth.

It is much easier to *"test"* this with static analysis, but it would require adding additional tools to the Docker image. Or the test could be a sort of plug-in for whatever compiler is being used, to traverse the AST of the code while building, and causing a failure when detecting this particular pattern.

Or you could just use a good ol' regular expression, `r"^.\w*size\w*(\w*)\w*>\w*0$"` and call it a day, but it is not the most reliable approach.

gavanderhoorn, 2018-11-05:
> See the following for an example: https://github.com/robust-rosin/robust/blob/master/mavros/0e2ea0c.bug

The bug description says:

> This commit performs two small performance/hygiene-related fixes to the code, discovered using the cppcheck static analyser.

Can't we do the same?

@git-afsantos wrote:
> It is much easier to "test" this with static analysis, but it would require adding additional tools to the Docker image.

Do you see that as a problem?

git-afsantos, 2018-11-06:
I do not see it as a problem at all, but what would we have to change in the build process to inject these additional *dependencies*?

> Can't we do the same?

We could, since it is clearly the best way to *detect* these issues. It just does not follow the traditional expectation/definition of what a test is.

It is really just a question of how flexible our methodology/ruleset is, to accept these alternative methods, as opposed to a standard unit/integration test. Maybe @wasowski can provide a more definitive answer.

gavanderhoorn, 2018-11-24:
Ping @wasowski?

wasowski, 2019-10-18:
Overtime, I got to belief that we should not use too much time for bug smells for the current data set.  Code smells are even more subtle to decide about than bugs. How smelly is smelly enough to include ? We are discussing very potential problems quite often.  I believe that we should refine our method to get a good sample set of code smells.  Seems out of scope now.  I vote for closing the issue.

gavanderhoorn, 2019-10-18:
I agree.

gavanderhoorn, 2019-10-18:
And I just did.

closed 2019-10-18 by gavanderhoorn