robust-rosin/robust #158: How to reproduce Turtlebot f01d952?

git-afsantos, 2018-10-31:
The bug is that a runtime dependency was missing in the `depth_image_proc` package XML. This caused an image processing nodelet to crash during runtime.

> [ INFO] [1402564815.530736554]: /camera/rgb/camera_info -> /camera/rgb/camera_info
[ERROR] [1402564815.727176562]: Failed to load nodelet [/camera/depth_metric_rect] of type
 [depth_image_proc/convert_metric]: Failed to load library /opt/ros/indigo/lib//libdepth_image_proc.so.
 Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that
 names are consistent between this macro and your XML. Error string: Could not load library (Poco
 exception = libeigen_conversions.so: cannot open shared object file: No such file or directory)
[FATAL] [1402564815.727410623]: Service call failed!

I was able to build the Docker image, and the test strategy seemed straightforward: just write a `.test` file that launches the nodelet, and it should crash in the buggy version.

I have done so, and it turns out that the nodelet does not crash.

Running `find .. -name "libeigen_conversions.so"` does find the library, within the container. Is this why the nodelet is not crashing? Even though the dependency is not declared, is the file still dynamically loaded?

Not sure how to proceed on this one.

gavanderhoorn, 2018-11-05:
> Running `find .. -name "libeigen_conversions.so"` does find the library, within the container. Is this why the nodelet is not crashing?

It's not actually crashing, not even in the original report: it's never even started, as the linker cannot find the library that contains some of the symbols that the nodelet needs.

In any case:

> Even though the dependency is not declared, is the file still dynamically loaded?

the dependency is not declared in the package manifest (`package.xml`), but that doesn't influence the way the GCC linker works: it'll just embed the info about `libeigen_conversions.so` in the nodelet binary (itself a `.so`).

If that nodelet is then loaded on a system that *does* have `libeigen_conversions.so`, everything will be fine. If `libeigen_conversions.so` is missing, you get the error message from the bug report.

The dependency declaration in the manifest is just for `rosdep` and other tools to make sure that the `eigen_conversions` package is present when setting up a ROS installation.

> I was able to build the Docker image, and the test strategy seemed straightforward: just write a `.test` file that launches the nodelet, and it should crash in the buggy version.
> 
> I have done so, and it turns out that the nodelet does not crash.

This most likely means that one of the other packages in the image *also* depends on `eigen_conversions` and thus it gets installed.

Why this didn't happen for the person who submitted the bug report, I don't know.

git-afsantos, 2018-11-05:
> It's not actually crashing, not even in the original report: it's never even started, as the linker cannot find the library that contains some of the symbols that the nodelet needs.

Hm, my bad, then. As I saw `[ERROR]` and `[FATAL]` in the reported error messages, I jumped to the conclusion that failing to load a nodelet would cause a crash of some sort.

> This most likely means that one of the other packages in the image _also_ depends on `eigen_conversions` and thus it gets installed.
> 
> Why this didn't happen for the person who submitted the bug report, I don't know.

Hm, then I will have to look further into it. If some other package also depends on `eigen_conversions`, and this package is itself a dependency (hence being installed in our process), this might make the bug unreproducible.

Now you got me thinking, do we use the buggy or fixed commit to determine dependencies? I would guess it is the buggy one, since we have been feeding issue report dates to the time machine.

If the Docker file used the fixed version to determine all dependencies, then this would explain `eigen_conversions` being there.

---

Off-topic: could bugs like this, where the fixed version has more dependencies than the buggy version, pose a problem to our current image build process?

gavanderhoorn, 2018-11-24:
> Off-topic: could bugs like this, where the fixed version has more dependencies than the buggy version, pose a problem to our current image build process?

well, sort of.

I think @ChrisTimperley captured this in #159.