robust-rosin/robust #161: build image for care-o-bot/105dc16

ipa-jfh, 2018-11-14:
Lives on in branch 105dc16:
I already created a `manifest.bugzoo.yml` and a `deps.rosinstall` for this bug.
https://github.com/robust-rosin/robust/blob/105dc16/care-o-bot/105dc16/

gavanderhoorn, 2018-11-14:
Then I'm not sure what you want me to do :)

See #137 for info on the "new way" of building images.

ipa-jfh, 2018-11-20:
@gavanderhoorn  do you know where this error might come from? It seems like version `0.5.5-0` does not exist. 

```bash
Step 31/43 : RUN wstool init -j8 ${ROS_WSPACE}/src ${ROS_WSPACE}/deps.rosinstall
 ---> Running in d2a08fd6714c
Using initial elements from: /ros_ws/deps.rosinstall
Writing /ros_ws/src/.rosinstall
...
[tf_conversions] Fetching https://github.com/ros-gbp/geometry-release/archive/release/indigo/tf_conversions/1.11.3-1.tar.gz (version geometry-release-release-indigo-tf_conversions-1.11.3-1) to /ros_ws/src/tf_conversions
[tf_conversions] Done.
[tf] Fetching https://github.com/ros-gbp/geometry-release/archive/release/indigo/tf/1.11.3-1.tar.gz (version geometry-release-release-indigo-tf-1.11.3-1) to /ros_ws/src/tf
[tf] Done.
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2_py-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2_py-0.5.5-0']"[/vcstools]

[tf2_py] Fetching https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_py/0.5.5-0.tar.gz (version geometry_experimental-release-release-indigo-tf2_py-0.5.5-0) to /ros_ws/src/tf2_py
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2_msgs-0.5.5-0']"[/vcstools]

[tf2_msgs] Fetching https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_msgs/0.5.5-0.tar.gz (version geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0) to /ros_ws/src/tf2_msgs
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2_ros-0.5.5-0']"[/vcstools]

[tf2_ros] Fetching https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.5-0.tar.gz (version geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0) to /ros_ws/src/tf2_ros
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2-0.5.5-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2-0.5.5-0']"[/vcstools]

[tf2] Fetching https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz (version geometry_experimental-release-release-indigo-tf2-0.5.5-0) to /ros_ws/src/tf2
[std_msgs,std_srvs,tf,tf2,tf2_msgs,tf2_py,tf2_ros,tf_conversions] still active
[urdf_parser_plugin] Fetching https://github.com/ros-gbp/robot_model-release/archive/release/indigo/urdf_parser_plugin/1.11.5-0.tar.gz (version robot_model-release-release-indigo-urdf_parser_plugin-1.11.5-0) to /ros_ws/src/urdf_parser_plugin
[urdf_parser_plugin] Done.
[trajectory_msgs] Fetching https://github.com/ros-gbp/common_msgs-release/archive/release/indigo/trajectory_msgs/1.11.4-0.tar.gz (version common_msgs-release-release-indigo-trajectory_msgs-1.11.4-0) to /ros_ws/src/trajectory_msgs
[trajectory_msgs] Done.
[visualization_msgs] Fetching https://github.com/ros-gbp/common_msgs-release/archive/release/indigo/visualization_msgs/1.11.4-0.tar.gz (version common_msgs-release-release-indigo-visualization_msgs-1.11.4-0) to /ros_ws/src/visualization_msgs
[visualization_msgs] Done.
[topic_tools] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/indigo/topic_tools/1.11.9-0.tar.gz (version ros_comm-release-release-indigo-topic_tools-1.11.9-0) to /ros_ws/src/topic_tools
[topic_tools] Done.
[urdf] Fetching https://github.com/ros-gbp/robot_model-release/archive/release/indigo/urdf/1.11.5-0.tar.gz (version robot_model-release-release-indigo-urdf-1.11.5-0) to /ros_ws/src/urdf
[urdf] Done.
[xmlrpcpp] Fetching https://github.com/ros-gbp/ros_comm-release/archive/release/indigo/xmlrpcpp/1.11.9-0.tar.gz (version ros_comm-release-release-indigo-xmlrpcpp-1.11.9-0) to /ros_ws/src/xmlrpcpp
[xmlrpcpp] Done.
[topic_tools,trajectory_msgs,urdf,urdf_parser_plugin,visualization_msgs,xmlrpcpp] still active
Exception caught during install: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /ros_ws/src/tf2 failed.
Error processing 'tf2_msgs' : [tf2_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_msgs/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_msgs-0.5.5-0 into /ros_ws/src/tf2_msgs failed.
Error processing 'tf2_py' : [tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_py/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_py-0.5.5-0 into /ros_ws/src/tf2_py failed.
Error processing 'tf2_ros' : [tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0 into /ros_ws/src/tf2_ros failed.
ERROR in config: Error processing 'tf2' : [tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.5-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.5-0 into /ros_ws/src/tf2 failed.
[... cut ...]
    version: geometry_experimental-release-release-indigo-tf2_py-0.5.5-0
- tar:
    local-name: tf2_ros
    uri: https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.5-0.tar.gz
    version: geometry_experimental-release-release-indigo-tf2_ros-0.5.5-0
- tar:
    local-name: tf_conversions
    uri: https://github.com/ros-gbp/geometry-release/archive/release/indigo/tf_conversions/1.11.3-1.tar.gz
    version: geometry-release-release-indigo-tf_conversions-1.11.3-1
```

gavanderhoorn, 2018-11-24:
Unassigning myself here as this should-just-work.

If it doesn't, please clarify.