robust-rosin/robust #162: Add support for bugs that span multiple repositories

ChrisTimperley, 2018-11-14:
The current Dockerfile assumes that there is a single PUT (and corresponding fork). This assumption is broken by a small number of bugs that occur in more than a single repository.

Related to #56.

gavanderhoorn, 2018-11-15:
We could consider using a `.rosinstall` file for the PUTs as well, in addition to the `deps.rosinstall` file.

That way we could still pass only a single arg to the Docker build command, and there could be any nr of PUTs in the `.rosinstall` file.

gavanderhoorn, 2018-11-17:
The time machine now supports multiple PUTs properly, so if `deps.rosinstall` is generated like so:

```
rosinstall_generator_tm.sh <some_time_stamp> --flat --tar --deps --deps-only pkg1 pkg2 > deps.rosinstall
```

it should contain the (transitive) dependencies of both `pkg1` and `pkg2` combined.

If we have a second `.rosinstall` file with entries for `pkg1` and `pkg2`, a workspace seeded with that last `.rosinstall` file should build cleanly in the image (provided it's not a build error of course).

ChrisTimperley, 2018-11-24:
> We could consider using a `.rosinstall` file for the PUTs as well, in addition to the `deps.rosinstall` file.
> 
> That way we could still pass only a single arg to the Docker build command, and there could be any nr of PUTs in the `.rosinstall` file.

I like this approach. We would need to figure out a way to update `fix` and `unfix`. One possibility is that we store a separate `.rosinstall` file for the buggy and fixed versions of the PUTs. This might also address missing dependency bugs? #159