robust-rosin/robust #164: Docker build fails for Turtlebot 891cb68

git-afsantos, 2018-11-14:
Docker command:

```bash
docker build -t robustrosin/test-891cb68 \
-f ../../Dockerfile \
--build-arg UBUNTU_VERSION=precise \
--build-arg IS_BUILD_FAILURE=no \
--build-arg USE_APT_OLD_RELEASES=no \
--build-arg USE_OSRF_REPOS=yes \
--build-arg ROS_DISTRO=hydro \
--build-arg CATKIN_PKG=turtlebot_bringup \
--build-arg REPO_FORK_URL=https://github.com/robust-rosin/turtlebot \
--build-arg REPO_BUG_COMMIT=981a09902f5b73a24dc63bbde3977b8c6750b583 \
--build-arg REPO_FIX_COMMIT=799db373131fcc022b6bda371a1884f9b23e7a7c \
. 2>&1 | tee build.log
```

Error message:

> **E: Unable to locate package libopenni-dev**
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y libopenni-dev] failed
executing command [apt-get install -y ftdi-eeprom]
executing command [apt-get install -y libeigen3-dev]
executing command [apt-get install -y libapr1-dev]
executing command [apt-get install -y libaprutil1-dev]
executing command [apt-get install -y libopenni-dev]
The command `/bin/sh -c apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*` returned a non-zero code: 1

<details>
<summary>Build log</summary>
Sending build context to Docker daemon  35.56kB

Step 1/43 : ARG UBUNTU_VERSION
Step 2/43 : FROM alpine:3.7 as fork
 ---> 3fd9065eaf02
Step 3/43 : ARG REPO_FORK_URL
 ---> Using cache
 ---> 7c79e9d9f9af
Step 4/43 : RUN apk --no-cache add git
 ---> Using cache
 ---> 8749c9079268
Step 5/43 : RUN echo "[ROBUST] cloning repo: '${REPO_FORK_URL}'"  && git clone "${REPO_FORK_URL}" /tmp/repo-under-test  && echo "[ROBUST] cloned repo."
 ---> Using cache
 ---> 117837942fd9
Step 6/43 : FROM ubuntu:${UBUNTU_VERSION}
 ---> 5b117edd0b76
Step 7/43 : ARG ROS_DISTRO
 ---> Using cache
 ---> 291a315e898e
Step 8/43 : ARG USE_APT_OLD_RELEASES
 ---> Using cache
 ---> 34640682334c
Step 9/43 : ARG CATKIN_PKG
 ---> Using cache
 ---> e6dfe641ed2b
Step 10/43 : ARG REPO_FIX_COMMIT
 ---> Using cache
 ---> e79ba249b09d
Step 11/43 : ARG REPO_BUG_COMMIT
 ---> Using cache
 ---> c964413685a3
Step 12/43 : ARG IS_BUILD_FAILURE
 ---> Using cache
 ---> 16b85331b0b1
Step 13/43 : ENV ROS_DISTRO "${ROS_DISTRO}"
 ---> Using cache
 ---> ef126f6c326b
Step 14/43 : RUN echo "[ROBUST]: building image for ROS_DISTRO: '${ROS_DISTRO}'"
 ---> Using cache
[... 897 lines cut ...]
Setting up libssl-dev (1.0.1-4ubuntu5.39) ...
Setting up libkrb5-dev (1.10+dfsg~beta1-2ubuntu0.7) ...
Setting up libpq-dev (9.1.24-0ubuntu0.12.04) ...
Setting up libmysqlclient-dev (5.5.54-0ubuntu0.12.04.1) ...
Setting up libaprutil1-dev (1.3.12+dfsg-3) ...
Setting up libssl-doc (1.0.1-4ubuntu5.39) ...
Processing triggers for libc-bin ...
ldconfig deferred processing now taking place
Reading package lists...
Building dependency tree...
Reading state information...
[91mE[0m[91m: [0m[91mUnable to locate package libopenni-dev[0m[91m
[0m[91mERROR: the following rosdeps failed to install[0m[91m
[0m[91m  apt: command [apt-get install -y libopenni-dev] failed[0m[91m
[0m[1mexecuting command [apt-get install -y ftdi-eeprom][0m
[1mexecuting command [apt-get install -y libeigen3-dev][0m
[1mexecuting command [apt-get install -y libapr1-dev][0m
[1mexecuting command [apt-get install -y libaprutil1-dev][0m
[1mexecuting command [apt-get install -y libopenni-dev][0m
The command '/bin/sh -c apt-get clean  && apt-get update  && rosdep init  && rosdep update  && rosdep install --from-paths src -i --rosdistro=${ROS_DISTRO} -y       --skip-keys="python-rosdep python-catkin-pkg python-rospkg"  && apt-get clean  && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1
</details>

Searching for [libopenni-dev](https://packages.ubuntu.com/search?keywords=libopenni-dev) yields nothing for Ubuntu Precise, indeed. Is this one of those OSRF repo cases? I provided the OSRF build argument to the command, but I ran this on master, and I am not sure whether the change has been merged yet.

@gavanderhoorn @ChrisTimperley

gavanderhoorn, 2018-11-14:
How did you generate the `.rosinstall` file?

Did you add the `--tar` option to `rosinstall_generator_tm.sh`? Because the output you quoted showed a lot of "cloning .." which I would not expect.

git-afsantos, 2018-11-14:
> Did you add the `--tar` option to `rosinstall_generator_tm.sh` (if using the new version of course)?

I am using it, and I did not. I figured it should make some difference, but the build was already in progress so I let it be. That probably explains why it is taking so long and building so many packages.

Guess I learned my lesson for the next one.

gavanderhoorn, 2018-11-14:
Not adding `--tar` will also bloat the resultant image quite a bit, so we should try to avoid that.

The `.rosinstall` file should probably be regenerated for this one.

git-afsantos, 2018-11-14:
Agreed. I am restarting the build with the proper `.rosinstall` now.

git-afsantos, 2018-11-14:
Even with `--tar` this build takes a *lot* of time. And now it failed with

> fatal: Couldn't find remote ref 799db373131fcc022b6bda371a1884f9b23e7a7c

I am going to `docker image prune` and pray it works the next time.

gavanderhoorn, 2018-11-17:
Could this be related to the issue that @ChrisTimperley is trying to fix in #149?

git-afsantos, 2018-11-17:
I think it is. I ran into similar issues before, and his dev branch seemed to fix it, but I have been working on master most of the time, so this often comes back to bite me.

git-afsantos, 2018-11-22:
So I just tried the newest fix from #149, and now the build is stuck on another issue:

> make: unrecognized option `--only-pkg-with-deps=turtlebot_bringup`
> (...)
> The command `/bin/sh -c echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]` returned a non-zero code: 1

<details>
<summary>Here follows the new build log.</summary>
Sending build context to Docker daemon  45.62kB

Step 1/43 : ARG UBUNTU_VERSION
Step 2/43 : FROM alpine:3.7 as fork
 ---> 3fd9065eaf02
Step 3/43 : ARG REPO_FORK_URL
 ---> Running in 552f85eb2fe2
Removing intermediate container 552f85eb2fe2
 ---> eda8b91de1a5
Step 4/43 : RUN apk --no-cache add git
 ---> Running in b39051862371
fetch http://dl-cdn.alpinelinux.org/alpine/v3.7/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.7/community/x86_64/APKINDEX.tar.gz
(1/6) Installing ca-certificates (20171114-r0)
(2/6) Installing libssh2 (1.8.0-r2)
(3/6) Installing libcurl (7.61.1-r1)
(4/6) Installing expat (2.2.5-r0)
(5/6) Installing pcre2 (10.30-r0)
(6/6) Installing git (2.15.3-r0)
Executing busybox-1.27.2-r7.trigger
Executing ca-certificates-20171114-r0.trigger
OK: 19 MiB in 17 packages
Removing intermediate container b39051862371
 ---> 8fddd8c5b14d
Step 5/43 : RUN echo "[ROBUST] cloning repo: '${REPO_FORK_URL}'"  && git clone "${REPO_FORK_URL}" /tmp/repo-under-test  && echo "[ROBUST] cloned repo."
 ---> Running in e8c706ef924d
[ROBUST] cloning repo: 'https://github.com/robust-rosin/turtlebot'
[91mCloning into '/tmp/repo-under-test'...
[0m[ROBUST] cloned repo.
Removing intermediate container e8c706ef924d
 ---> 9ccca2c377ee
Step 6/43 : FROM ubuntu:${UBUNTU_VERSION}
 ---> 5b117edd0b76
Step 7/43 : ARG ROS_DISTRO
 ---> Using cache
 ---> 291a315e898e
Step 8/43 : ARG USE_APT_OLD_RELEASES
 ---> Using cache
 ---> 34640682334c
Step 9/43 : ARG CATKIN_PKG
 ---> Using cache
 ---> e6dfe641ed2b
Step 10/43 : ARG REPO_FIX_COMMIT
 ---> Using cache
 ---> e79ba249b09d
Step 11/43 : ARG REPO_BUG_COMMIT
 ---> Using cache
 ---> c964413685a3
Step 12/43 : ARG IS_BUILD_FAILURE
 ---> Using cache
 ---> 16b85331b0b1
Step 13/43 : ENV ROS_DISTRO "${ROS_DISTRO}"
 ---> Using cache
 ---> ef126f6c326b
Step 14/43 : RUN echo "[ROBUST]: building image for ROS_DISTRO: '${ROS_DISTRO}'"
 ---> Using cache
 ---> 7a93300343a0
Step 15/43 : ENV ROS_WSPACE=/ros_ws
 ---> Using cache
 ---> e82504f466fb
Step 16/43 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
[... 170 lines cut ...]
  -t, --touch                 Touch targets instead of remaking them.
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.
[0m[91m
This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
[0mBase path: /ros_ws
Source space: /ros_ws/src
Build space: /ros_ws/build
Devel space: /ros_ws/devel
Install space: /ros_ws/install
Creating symlink "/ros_ws/src/CMakeLists.txt" pointing to "/opt/ros/hydro/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /ros_ws/src -DCATKIN_DEVEL_PREFIX=/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/ros_ws/install" in "/ros_ws/build"
####
####
#### Running command: "make --only-pkg-with-deps=turtlebot_bringup -j1 -l1" in "/ros_ws/build"
####
[91mInvoking "make" failed[0m[91m
[0mThe command '/bin/sh -c echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]' returned a non-zero code: 1
</details>

gavanderhoorn, 2018-11-22:
Looks like it's not running `catkin_make` but `make`. A copy-pasta somewhere @ChrisTimperley ?

git-afsantos, 2018-11-22:
Same thing happens on `master`.

gavanderhoorn, 2018-11-24:
> Is not `--only-pkg-with-deps` available in Hydro? Or has something else changed in the build process?

I only now saw your edit.

It could well be that the version of `catkin` is affecting this.

`--only-pkg-with-deps` was added in [0.5.78](https://github.com/ros/catkin/blob/928c2025206e1a3f0d9d504ef5b26efe1cc95d8b/CHANGELOG.rst#0578-2014-01-07), so if the version being used is older, that would be a problem.

But I believe we're always installing the *latest* `catkin-pkg` ([here](https://github.com/robust-rosin/robust/blob/dc3bf55c90ae7fe5257df1178413676d4a4522b2/Dockerfile#L95-L101)), so that should not happen.

git-afsantos, 2018-11-24:
Then I should have a look at the build log, just to make sure that the catkin version is greater than that.

git-afsantos, 2018-11-24:
Yup, just opened the build log above and did a Ctrl+F for *catkin*. Found this line, when catkin is invoked.

```
catkin 0.5.77
```

gavanderhoorn, 2018-11-24:
Interesting.

I don't really understand right now how that old a version of Catkin gets installed though.

gavanderhoorn, 2018-11-24:
Hm. Maybe because it's using a cached layer?

```
Step 25/43 : RUN pip install --upgrade       wheel       rosdep       wstool       rosinstall       rospkg       catkin_pkg  && apt-get clean  && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 043d5af32a80
```

git-afsantos, 2018-11-24:
That might be the most plausible explanation, and also why I am seeing it in more than one bug (and why nobody else reported it).

I should probably get rid of that cache and try again.

gavanderhoorn, 2018-11-24:
Well .. we should probably find a way to force Docker to not use a cached layer there.

But I still don't understand how that old version of `catkin_pkg` got installed in the first place. The current version is `0.6.16`.

ChrisTimperley, 2018-11-24:
> Hm. Maybe because it's using a cached layer?
> 
> ```
> Step 25/43 : RUN pip install --upgrade       wheel       rosdep       wstool       rosinstall       rospkg       catkin_pkg  && apt-get clean  && rm -rf /var/lib/apt/lists/*
>  ---> Using cache
>  ---> 043d5af32a80
> ```

If we know that `catkin_pkg` needs to be a particular version, then we change `catkin_pkg` to `catkin_pkg>=x.x.x`.