robust-rosin/robust #171: Bugzoo looks for the Dockerfile at the wrong location

ipa-jfh, 2018-11-20:
Bugzoo looks for the Dockerfile at the wrong location (correct location is inside the robust root).
If I have the `.bugzoo.yml` (generated by script) in the robust root it looks in two folders below (../.../).
And if I have the file in my bug folder it looks inside the bug folder. Thus, somehow it is interchanged. 

Surprisingly `$ bugzoo bug build --force robust:0af76d5` still works.

### My robust_bugzoo.yml:

```yaml
blueprints:
- arguments:
    CATKIN_PKG: cob_control
    IS_BUILD_FAILURE: 'yes'
    REPO_BUG_COMMIT: 105dc16_robust_buggy
    REPO_FIX_COMMIT: 105dc16_robust_fixed
    REPO_FORK_URL: https://github.com/robust-rosin/cob_control
    ROS_DISTRO: indigo
    UBUNTU_VERSION: 14.04.5
    USE_APT_OLD_RELEASES: false   # <-- this looks also not correct. Expected "yes"
    USE_OSRF_REPOS: false   # <-- this looks also not correct. Expected "yes"
  context: care-o-bot/105dc16
  file: Dockerfile
  tag: robustrosin/robust:105dc16
bugs:
- compiler:
    time-limit: 300
    type: catkin
    workspace: /ros_ws/src
  dataset: robust
  image: robustrosin/robust:105dc16
  languages:
  - cpp
  name: robust:105dc16
  program: cob_control
  source-location: /ros_ws/src
  test-harness:
    type: empty
version: '1.0'
```

### If robust_bugzoo.yml is in the root (robust/robust_bugzoo.yml):

```bash
$ bugzoo bug build --force robust:105dc16
building bug: robust:105dc16
Building image: robustrosin/robust:105dc16
Traceback (most recent call last):
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 104, in build
    shutil.copy(instructions.file_abs, tf)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/shutil.py", line 235, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/shutil.py", line 114, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jfh/stuff_for_ros/robust/../../Dockerfile'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jfh/stuff_for_ros/bugzoo_venv/bin/bugzoo", line 11, in <module>
    sys.exit(main())
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 541, in main
    args.func(args)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 489, in <lambda>
    cmd.set_defaults(func=lambda args: build_bug(rbox, args.bug, args.force))
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 103, in build_bug
    rbox.bugs.build(bug, force=force)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/bug.py", line 82, in build
    quiet=quiet)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 130, in build
    os.remove(tf)
FileNotFoundError: [Errno 2] No such file or directory: '/home/jfh/stuff_for_ros/robust/.Dockerfile'
```
I can make it work if I copy the `Dockerfile` to `robust/../../Dockerfile`.

### If robust_bugzoo.yml is in the bug folder (robust/care-o-bot/105dc16/robust_bugzoo.yml):

```bash
$ bugzoo bug build --force robust:105dc16
building bug: robust:105dc16
Building image: robustrosin/robust:105dc16
Traceback (most recent call last):
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 104, in build
    shutil.copy(instructions.file_abs, tf)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/shutil.py", line 235, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/shutil.py", line 114, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jfh/stuff_for_ros/robust/care-o-bot/105dc16/Dockerfile'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jfh/stuff_for_ros/bugzoo_venv/bin/bugzoo", line 11, in <module>
    sys.exit(main())
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 541, in main
    args.func(args)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 489, in <lambda>
    cmd.set_defaults(func=lambda args: build_bug(rbox, args.bug, args.force))
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 103, in build_bug
    rbox.bugs.build(bug, force=force)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/bug.py", line 82, in build
    quiet=quiet)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 130, in build
    os.remove(tf)
FileNotFoundError: [Errno 2] No such file or directory: '/home/jfh/stuff_for_ros/robust/care-o-bot/105dc16/care-o-bot/105dc16/.Dockerfile'
```

ipa-jfh, 2018-11-20:
Also:
If I use the auto-generated `.bugzoo.yml` Bugzoo looks for the `deps.rosinstall` file in the robust root, but it should look inside the bug's folder.
(If I copy it manually to the robust root it works)

```bash
Step 30/43 : COPY deps.rosinstall .
Traceback (most recent call last):
  File "/home/jfh/stuff_for_ros/bugzoo_venv/bin/bugzoo", line 11, in <module>
    sys.exit(main())
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 541, in main
    args.func(args)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 489, in <lambda>
    cmd.set_defaults(func=lambda args: build_bug(rbox, args.bug, args.force))
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 103, in build_bug
    rbox.bugs.build(bug, force=force)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/bug.py", line 82, in build
    quiet=quiet)
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 124, in build
    raise ImageBuildFailed(name, log)
bugzoo.exceptions.ImageBuildFailed: failed to build Docker image: robustrosin/robust:105dc16

```

ChrisTimperley, 2018-11-20:
Just to confirm: there is only one `.bugzoo.yml` file in your ROBUST repo at any given time, right?

ipa-jfh, 2018-11-20:
My bad. There was another one in the root - probably this one caused the issue. I will have a look on it again tomorrow.

Thank you for your response!

gavanderhoorn, 2018-11-24:
Should we close this, or should the BugZoo scripts (or the ones here in this repository) be somehow extended to catch this sort of situation?