robust-rosin/robust #172: Update build-rosinstall.py to support new time machine

ChrisTimperley, 2018-11-20:


gavanderhoorn, 2018-11-20:
Sorry, haven't finished it yet .. :(

ChrisTimperley, 2018-11-20:
> Sorry, haven't finished it yet .. :(

Ah, no worries. Do you expect the interface to be any different to the one used by https://github.com/rosin-project/rosinstall_generator_time_machine/pull/8?

gavanderhoorn, 2018-11-20:
No.

Or at least not significantly.

If/when the Python version gets here the ros distro arg might need a `--rosdistro=` prepended and the timestamp a `--timestamp`, but that's about it I think.

ChrisTimperley, 2018-11-20:
> No.
> 
> Or at least not significantly.
> 
> If/when the Python version gets here the ros distro arg might need a `--rosdistro=` prepended and the timestamp a `--timestamp`, but that's about it I think.

Great. Do you mind if I go ahead and update `build-bugzoo.py` to support https://github.com/rosin-project/rosinstall_generator_time_machine/pull/8?

gavanderhoorn, 2018-11-20:
No, go ahead.

As long as it's clear that people need the PR. And not the current `master`.

Perhaps you could make `build-bugzoo.py` check for presence of the Docker image the new version uses?

gavanderhoorn, 2018-11-24:
Time machine was upgraded, so `master` branch of that repository now contains the new version.

Docker image is now also v`03`.

commit f61ebc84 (2018-11-24): Updated build-rosinstall.py to time machine v2 (fixes #172, fixes #150) (#173)

* fixed dir_bug for new directory structure

* remove dir creation code

* added Docker image checking

* check for existence of binary

* added requirements.txt

* updated time machine command

* updated script execution

* write contents to deps.rosinstall

* added repository renaming fix

* removed unnecessary vars

* added handling of GitHub issues

* multiple ROS packages are now supported

* handle stdin, stderr and stdout

* add support for missing deps (fixes #150)

* described missing-dependencies

* prefe [... cut]
closed 2018-11-24 by ChrisTimperley