robust-rosin/robust #174: Quickest way to find ROS distribution?

git-afsantos, 2018-11-22:
It might be silly to come up with this question at this point, but I am just wondering whether there is a way to simplify this that I may be missing.

I have been looking at commit dates and package version numbers, to try and map that into the corresponding ROS distro. It works, most of the time, but it feels like guessing.

Is there anything in the packages/repos that could easily tell the target ROS distro?

git-afsantos, 2018-11-24:
So, basically, working out version numbers or hoping the related issue/PR have some hints. I am glad that I have not missed something obvious, but I am also disheartened that this information is not mandatory (e.g. in `package.xml`).

I guess this explains why there is some debate over how ROS repositories and branches should be organised.

gavanderhoorn, 2018-11-24:
>  I am also disheartened that this information is not mandatory (e.g. in `package.xml`).

How would you see `package.xml` helping here?

gavanderhoorn, 2018-11-24:
One of the "issues" here is that packages with the same version can be released into multiple ROS distributions. In those cases, there is no 1-to-1 mapping.

> So, basically, working out version numbers or hoping the related issue/PR have some hints

I think you make it sound much more like guessing than it actually is: commits are linked to tags in a 1-to-1 way. Those tags correspond to package releases 1-to-1. At that point you need to look in `rosdistro` and see when that release was submitted (by Bloom or manually).

git-afsantos, 2018-11-24:
> How would you see `package.xml` helping here?

It requires a number of fields already. Adding one more for the *intended* ROS distro (or the *minimum required* distro) of that package/version makes some sense to me.

> commits are linked to tags in a 1-to-1 way

Yes, I have been looking at those in most cases.

Just out of curiosity, this is sounding like something that could be automated, *a la* `get_issue_creation_date`.

gavanderhoorn, 2018-11-24:
> It requires a number of fields already. Adding one more for the _intended_ ROS distro (or the _minimum required_ distro) of that package/version makes some sense to me.

Well as I already wrote: the same package can be released into multiple ROS distributions. So such a field would (apart from increasing maintenance) lose its meaning pretty quickly I believe.

git-afsantos, 2018-11-24:
It could be a list of distributions, then! :grin: 
But I see what you mean.

In any case, I think there is not much more to discuss here, and your edit above is already pretty useful, so this issue can be closed, unless someone else wants to chime in.

gavanderhoorn, 2018-11-24:
Let's wait and see whether @ChrisTimperley has an idea that makes this easier.

git-afsantos, 2018-11-28:
I think this can be closed for now. :+1:

closed 2018-11-28 by git-afsantos