robust-rosin/robust #175: make: unrecognized option '--only-pkg-with-deps'

git-afsantos, 2018-11-22:
I came across this error in https://github.com/robust-rosin/robust/issues/164#issuecomment-440971623, and now I am getting the same thing when trying to build Kobuki `fd6b589`.

With the latter, since it `IS_BUILD_FAILURE`, the issue only manifests when running the container.

I am on the latest versions of `master` for robust and the time machine v2 PR.

I am starting to suspect it might be something wrong with my setup.
@gavanderhoorn @ChrisTimperley

gavanderhoorn, 2018-11-24:
Just copying my https://github.com/robust-rosin/robust/issues/164#issuecomment-441379754:

> > Is not `--only-pkg-with-deps` available in Hydro? Or has something else changed in the build process?
> 
> I only now saw your edit.
> 
> It could well be that the version of `catkin` is affecting this.
> 
> `--only-pkg-with-deps` was added in [0.5.78](https://github.com/ros/catkin/blob/928c2025206e1a3f0d9d504ef5b26efe1cc95d8b/CHANGELOG.rst#0578-2014-01-07), so if the version being used is older, that would be a problem.
> 
> But I believe we're always installing the _latest_ `catkin-pkg` ([here](https://github.com/robust-rosin/robust/blob/dc3bf55c90ae7fe5257df1178413676d4a4522b2/Dockerfile#L95-L101)), so that should not happen.

git-afsantos, 2018-11-24:
Also copying my other https://github.com/robust-rosin/robust/issues/164#issuecomment-441380401

> 
> 
> Yup, just opened the build log above and did a Ctrl+F for _catkin_. Found this line, when catkin is invoked.
> 
> ```
> catkin 0.5.77
> ```

It seems that sometimes catkin 0.5.78 might not be available. Now to dig up the cause.

git-afsantos, 2018-11-24:
My build for Kobuki `fd6b589` is using catkin `0.5.73`.

gavanderhoorn, 2018-11-25:
I believe https://github.com/robust-rosin/robust/issues/177#issuecomment-441449277 is also relevant here:

> > [catkin] Fetching https://github.com/ros-gbp/catkin-release/archive/release/hydro/catkin/0.5.73-0.tar.gz (version catkin-release-release-hydro-catkin-0.5.73-0) to /ros_ws/src/catkin
> 
> Ah, right. I'd forgotten that `catkin` is actually a package in the ROS workspace. That's why old versions are being used.
> 
> Ok. So `--only-pkg-with-deps` cannot just be expected to always work.
> 
> Two options I believe:
> 
>     1. see if using `CATKIN_WHITELIST_PACKAGES` directly (that is essentially what `--only-pkg-with-deps` does
> 
>     2. always install a recent version of `catkin`
> 
>     3. manually `rm` or `CATKIN_IGNORE` any pkgs in the source repository of the PUT(s) that are not the PUT(s)
> 
> 
> Of these I think option 1 would be the best to check first: it looks like it's been supported since at least 2012 ([here](https://github.com/ros/catkin/blame/928c2025206e1a3f0d9d504ef5b26efe1cc95d8b/cmake/catkin_workspace.cmake#L25)) and should not be too difficult to implement / have the least amount of side-effects.

gavanderhoorn, 2018-11-25:
See the rest of the comments following https://github.com/robust-rosin/robust/issues/177#issuecomment-441449277 for a discussion on how to address the main problem raised in this issue.

ChrisTimperley, 2019-03-26:
#264 uses `catkin_tools` rather than `catkin_make`.

gavanderhoorn, 2019-03-26:
So, `catkin_tools` isn't included in the `.rosinstall` file, and will use the latest version as it's installed using `pip`.

`catkin_make` however is provided by `catkin`, which *is* part of the `.rosinstall` files.

That makes the build tool used be from the same point in time as the PUT and all its dependencies (`catkin` is essentially part of the dependency tree of the PUT(s)).

We could go with `catkin_tools` everywhere, but that would mean we're not replicating the state of the build tool at the time the bug was reported, and I believe we need to make that a concious decision.

ChrisTimperley, 2019-03-26:
> We could go with `catkin_tools` everywhere, but that would mean we're not replicating the state of the build tool at the time the bug was reported, and I believe we need to make that a concious decision.

I don't really see any tangible disadvantage here. We're only interested in reproducing the bug, and few (if any) bugs rely on the existence of a particular version of `catkin_make` in order to manifest. We would only need to deal a single consistent interface for `catkin`, rather than numerous versions of `catkin_make` with subtly different options and semantics.

gavanderhoorn, 2019-03-26:
That's not necessarily true: `catkin_tools` has still not even seen an alpha-release, and is not considered mature enough for "production" use. It had/has quite a few serious issues, some of which were actually only resolved *today*.

`catkin` however has seen much more development and testing, by a much larger userbase.

ChrisTimperley, 2019-03-26:
> That's not necessarily true: `catkin_tools` has still not even seen an alpha-release, and is not considered mature enough for "production" use. It had/has quite a few serious issues, some of which [...]

Interesting. In that case, how do you feel about using a single version of `catkin` for all bugs?

Also, if we go with `catkin`, we'll need to figure out how to port this from `catkin_tools`:
https://github.com/robust-rosin/robust/pull/264/files#diff-3254677a7917c6c01f55212f86c57fbfR182

ChrisTimperley, 2019-03-26:
> Also, if we go with `catkin`, we'll need to figure out how to port this from `catkin_tools`:
> https://github.com/robust-rosin/robust/pull/264/files#diff-3254677a7917c6c01f55212f86c57fbfR182

It should be possible to build specific PUTs by passing a whitelist to `catkin_make`:

```
catkin_make -DCATKIN_WHITELIST_PACKAGES="package1;package2"
```

(reference: http://wiki.ros.org/catkin/commands/catkin_make)

gavanderhoorn, 2019-03-26:
re: whitelist: see https://github.com/robust-rosin/robust/issues/177#issuecomment-441449277.

ChrisTimperley, 2019-03-26:
> re: whitelist: see [#177 (comment)](https://github.com/robust-rosin/robust/issues/177#issuecomment-441449277).

Ah, I see. So we would need to implement the necessary logic to pass the recursive dependencies to `-DCATKIN_WHITELIST_PACKAGES`. I'll figure something out.

gavanderhoorn, 2019-03-26:
Let's not complicate things too much for ourselves.

If `catkin_tools` works, we could consider using it.

But as I wrote earlier: we should make a decision. It should not be a silent migration in "some" PR.

ChrisTimperley, 2019-03-26:
> Let's not complicate things too much for ourselves.
> 
> If `catkin_tools` works, we could consider using it.

Well, if we can build using `catkin_make --only-pkg-with-deps`, then using a fixed version of the more stable `catkin` (rather than `catkin_tools`) seems like a reasonable solution?

> But as I wrote earlier: we should make a decision. It should not be a silent migration in "some" PR.

Perhaps I wasn't clear enough above -- I agree that a decision needs to be made, and I wasn't planning to merge anything without a discussion.

commit bf58c939 (2019-04-04): Ensure catkin version >= 0.5.78 (adds --only-pkg-with-deps) (fixes #175) (#272)

* extract catkin version

* neater header

* update version and URI

* bug fix: <= to <

* added requirements.txt

* remove requirements.txt

* added packaging to scripts/requirements.txt

* use packaging

* tidy code
closed 2019-04-04 by ChrisTimperley