robust-rosin/robust #177: Docker build fails for Kobuki b213df3

git-afsantos, 2018-11-25:
https://github.com/robust-rosin/robust/blob/master/kobuki/b213df3/b213df3.bug

With errors:

> -- +++ processing catkin package: 'kobuki_auto_docking'
> -- ==> add_subdirectory(repo-under-test/kobuki_auto_docking)
> **CMake Error at /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:72 (find_package):
  Could not find a configuration file for package kdl_conversions.**
> [...]
> **CMake Error at /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:72 (find_package):
  Could not find a configuration file for package kobuki_dock_drive.**
> [...]
> -- +++ processing catkin package: 'kobuki_description'
> -- ==> add_subdirectory(repo-under-test/kobuki_description)
> **CMake Error at /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:72 (find_package):
  Could not find a configuration file for package xacro.**
> [...]
> -- +++ processing catkin package: 'kobuki_node'
> -- ==> add_subdirectory(repo-under-test/kobuki_node)

I built `deps.rosinstall` as per usual:

```bash
./rosinstall_generator_tm.sh "2013-08-30T09:46:00+00:00" hydro --deps --deps-only --tar kobuki_node > deps.rosinstall
```

The missing dependencies do not even seem to be related to the PUT (`kobuki_node`), but rather to other packages (within the same metapackage, I think).

gavanderhoorn, 2018-11-25:
`kobuki_dock_drive` and `kdl_conversions` are dependencies of `kobuki_auto_docking` (see [the wiki page](http://wiki.ros.org/kobuki_auto_docking?distro=hydro) and expand the *Dependencies* section of the *Package Header* box (to the left)).

Could you include the `.rosinstall` file in a comment here?

git-afsantos, 2018-11-25:
<details>
<summary>`deps.rosinstall`</summary>
- tar:
    local-name: actionlib
    uri: https://github.com/ros-gbp/actionlib-release/archive/release/hydro/actionlib/1.10.3-0.tar.gz
    version: actionlib-release-release-hydro-actionlib-1.10.3-0
- tar:
    local-name: actionlib_msgs
    uri: https://github.com/ros-gbp/common_msgs-release/archive/release/hydro/actionlib_msgs/1.10.2-0.tar.gz
    version: common_msgs-release-release-hydro-actionlib_msgs-1.10.2-0
- tar:
    local-name: angles
    uri: https://github.com/ros-gbp/geometry_angles_utils-release/archive/release/hydro/angles/1.9.9-0.tar.gz
    version: geometry_angles_utils-release-release-hydro-angles-1.9.9-0
- tar:
    local-name: bond
    uri: https://github.com/ros-gbp/bond_core-release/archive/release/hydro/bond/1.7.13-0.tar.gz
    version: bond_core-release-release-hydro-bond-1.7.13-0
- tar:
    local-name: bondcpp
    uri: https://github.com/ros-gbp/bond_core-release/archive/release/hydro/bondcpp/1.7.13-0.tar.gz
    version: bond_core-release-release-hydro-bondcpp-1.7.13-0
- tar:
    local-name: catkin
    uri: https://github.com/ros-gbp/catkin-release/archive/release/hydro/catkin/0.5.73-0.tar.gz
    version: catkin-release-release-hydro-catkin-0.5.73-0
- tar:
    local-name: class_loader
    uri: https://github.com/ros-gbp/class_loader-release/archive/release/hydro/class_loader/0.2.3-0.tar.gz
    version: class_loader-release-release-hydro-class_loader-0.2.3-0
- tar:
    local-name: cmake_modules
    uri: https://github.com/ros-gbp/cmake_modules-release/archive/release/hydro/cmake_modules/0.1.0-0.tar.gz
    version: cmake_modules-release-release-hydro-cmake_modules-0.1.0-0
- tar:
    local-name: cmd_vel_mux
    uri: https://github.com/yujinrobot-release/yujin_ocs-release/archive/release/hydro/cmd_vel_mux/0.4.0-0.tar.gz
    version: yujin_ocs-release-release-hydro-cmd_vel_mux-0.4.0-0
- tar:
    local-name: console_bridge
    uri: https://github.com/ros-gbp/console_bridge-release/archive/release/hydro/console_bridge/0.2.4-1.tar.gz
    version: console_bridge-release-release-hydro-console_bridge-0.2.4-1
- tar:
    local-name: cpp_common
    uri: https://github.com/ros-gbp/roscpp_core-release/archive/release/hydro/cpp_common/0.3.16-0.tar.gz
    version: roscpp_core-release-release-hydro-cpp_common-0.3.16-0
- tar:
    local-name: diagnostic_msgs
    uri: https://github.com/ros-gbp/common_msgs-release/archive/release/hydro/diagnostic_msgs/1.10.2-0.tar.gz
    version: common_msgs-release-release-hydro-diagnostic_msgs-1.10.2-0
- tar:
    local-name: diagnostic_updater
    uri: https://github.com/ros-gbp/diagnostics-release/archive/release/hydro/diagnostic_updater/1.8.0-0.tar.gz
    version: diagnostics-release-release-hydro-diagnostic_updater-1.8.0-0
- tar:
    local-name: dynamic_reconfigure
    uri: https://github.com/ros-gbp/dynamic_reconfigure-release/archive/release/hydro/dynamic_reconfigure/1.5.33-0.tar.gz
    version: dynamic_reconfigure-release-release-hydro-dynamic_reconfigure-1.5.33-0
- tar:
    local-name: ecl_build
    uri: https://github.com/yujinrobot-release/ecl_tools-release/archive/release/hydro/ecl_build/0.60.0-2.tar.gz
    version: ecl_tools-release-release-hydro-ecl_build-0.60.0-2
- tar:
    local-name: ecl_command_line
    uri: https://github.com/yujinrobot-release/ecl_core-release/archive/release/hydro/ecl_command_line/0.60.5-0.tar.gz
    version: ecl_core-release-release-hydro-ecl_command_line-0.60.5-0
- tar:
    local-name: ecl_concepts
    uri: https://github.com/yujinrobot-release/ecl_core-release/archive/release/hydro/ecl_concepts/0.60.5-0.tar.gz
    version: ecl_core-release-release-hydro-ecl_concepts-0.60.5-0
[... 332 lines cut ...]
- tar:
    local-name: urdfdom
    uri: https://github.com/ros-gbp/urdfdom-release/archive/release/hydro/urdfdom/0.2.8-1.tar.gz
    version: urdfdom-release-release-hydro-urdfdom-0.2.8-1
- tar:
    local-name: urdfdom_headers
    uri: https://github.com/ros-gbp/urdfdom_headers-release/archive/release/hydro/urdfdom_headers/0.2.3-1.tar.gz
    version: urdfdom_headers-release-release-hydro-urdfdom_headers-0.2.3-1
- tar:
    local-name: visualization_msgs
    uri: https://github.com/ros-gbp/common_msgs-release/archive/release/hydro/visualization_msgs/1.10.2-0.tar.gz
    version: common_msgs-release-release-hydro-visualization_msgs-1.10.2-0
- tar:
    local-name: xmlrpcpp
    uri: https://github.com/ros-gbp/ros_comm-release/archive/release/hydro/xmlrpcpp/1.9.48-0.tar.gz
    version: ros_comm-release-release-hydro-xmlrpcpp-1.9.48-0
- tar:
    local-name: yocs_controllers
    uri: https://github.com/yujinrobot-release/yujin_ocs-release/archive/release/hydro/yocs_controllers/0.4.0-0.tar.gz
    version: yujin_ocs-release-release-hydro-yocs_controllers-0.4.0-0
- tar:
    local-name: yocs_velocity_smoother
    uri: https://github.com/yujinrobot-release/yujin_ocs-release/archive/release/hydro/yocs_velocity_smoother/0.4.0-0.tar.gz
    version: yujin_ocs-release-release-hydro-yocs_velocity_smoother-0.4.0-0
</details>

git-afsantos, 2018-11-25:
The packages it complains about are indeed absent of the `.rosinstall`. But it only complains because it is trying to build `kobuki_auto_docking` and `kobuki_description`, which are *not* dependencies of the PUT, but belong to the same metapackage.

https://github.com/yujinrobot/kobuki/tree/b213df3324cd933ddfc1bbd7ef8dce50cc233d51

git-afsantos, 2018-11-25:
Building a `.rosinstall` for the whole `kobuki` metapackage might silence the issue, but does not sound like a real solution.

gavanderhoorn, 2018-11-25:
Is the white listing failing for some reason?

The [fork](https://github.com/robust-rosin/kobuki/tree/4a8fdfc23906dbc4b8984ad8df408df1cb59bf85) of `kobuki` we use of course has all the packages, including `kobuki_auto_docking`.

If the pkg white listing isn't working correctly, it would make sense for Catkin to try and build *everything*, which will fail, as, as you also observed, the `.rosinstall` file doesn't contain the dependencies for all those other nodes.

Can you post the complete build log?

gavanderhoorn, 2018-11-25:
O and please state the versions of both `robust-rosin/robust` and the time machine you are using.

gavanderhoorn, 2018-11-25:
> [catkin] Fetching https://github.com/ros-gbp/catkin-release/archive/release/hydro/catkin/0.5.73-0.tar.gz (version catkin-release-release-hydro-catkin-0.5.73-0) to /ros_ws/src/catkin

Ah, right. I'd forgotten that `catkin` is actually a package in the ROS workspace. That's why old versions are being used.

Ok. So `--only-pkg-with-deps` cannot just be expected to always work.

Two options I believe:

 1. see if using `CATKIN_WHITELIST_PACKAGES` directly (that is essentially what `--only-pkg-with-deps` does
 1. always install a recent version of `catkin`
 1. manually `rm` or `CATKIN_IGNORE` any pkgs in the source repository of the PUT(s) that are not the PUT(s)

Of these I think option 1 would be the best to check first: it looks like it's been supported since at least 2012 ([here](https://github.com/ros/catkin/blame/928c2025206e1a3f0d9d504ef5b26efe1cc95d8b/cmake/catkin_workspace.cmake#L25)) and should not be too difficult to implement / have the least amount of side-effects.

git-afsantos, 2018-11-25:
> see if using CATKIN_WHITELIST_PACKAGES directly (that is essentially what --only-pkg-with-deps does

This requires editing the part where `build.sh` is generated, right?

gavanderhoorn, 2018-11-25:
Yes, [this bit](https://github.com/robust-rosin/robust/blob/16ec82e60bf2fb4e204a20c13c3546056cfa0d64/Dockerfile#L182-L185).

You'd add something like: `-DCATKIN_WHITELIST_PACKAGES=foo;bar;fop` instead of the `--only-pkg-with-deps`.

gavanderhoorn, 2018-11-25:
@gavanderhoorn wrote:
> One disadvantage of this is that `--only-pkg-with-deps` also figures out what the *dependencies* are of the pkg that you pass it and also adds *those* to the whitelist. That would now become our responsibility.
> 
> For pkgs that essentially only depend on the pkgs in the underlay workspace (ie: those from the `.rosinstall` file) this would not be a problem (as they have no other dependencies present in the overlay workspace), but that may not always be true.

That's basically [this bit](https://github.com/ros/catkin/blob/e401545801d261aab46bfb4d93f5981b8acd2dc3/bin/catkin_make#L140-L159) of code.

git-afsantos, 2018-11-25:
Using `-DCATKIN_WHITELIST_PACKAGES=${CATKIN_PKG}` worked in this case. The issue now is to somehow emulate that bit in older catkin versions.

gavanderhoorn, 2018-11-25:
> Using `-DCATKIN_WHITELIST_PACKAGES=${CATKIN_PKG}` worked in this case. The issue now is to somehow emulate that bit in older catkin versions.

I don't understand. The point of using this is that it works in Catkin versions back to at least 2012.

gavanderhoorn, 2018-11-25:
It's only a few lines of Python and fairly stand-alone. We could either patch older versions of Catkin with it -- after it has been cloned into the underlay workspace -- or find a way to run it without being part of Catkin. The only thing we need from it is that list of pkgs.

gavanderhoorn, 2018-11-25:
[This](https://github.com/ros/catkin/commit/913488427d2ff18b808764d1eaf38acead67e18f) is the commit in which `--only-pkg-with-deps` was added. That seems to also have introduced `get_package_names_with_recursive_dependencies(..)`, which is unfortunate, as that is the main piece of functionality we'd like to re-use.

The good news is though that it seems like the infrastructure for `--only-pkg-with-deps` seems fairly stand-alone (note that the code that currently handles that option ([this](https://github.com/ros/catkin/blob/e401545801d261aab46bfb4d93f5981b8acd2dc3/bin/catkin_make#L140-L159)) seems a tad more complete).

gavanderhoorn, 2019-05-29:
I think this should now work as @ChrisTimperley overrides the version of `catkin` when it is lower than what we need (#272).

You'd need to rebase on current `master` @git-afsantos.