robust-rosin/robust #184: PUT fails to build for mavros/1a746c3

ChrisTimperley, 2018-11-27:
```
Step 41/43 : RUN echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]

 ---> Running in 908518245f1e
[ROBUST] attempting to build PUT...
[ROBUST] is a build failure expected? no.
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /ros_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: libmavconn;mavros;mavros_msgs
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - libmavconn
-- ~~  - mavros_msgs
-- ~~  - mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(repo-under-test/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- Performing Test COMPILER_SUPPORTS_STD_CXX14
-- Performing Test COMPILER_SUPPORTS_STD_CXX14 - Failed
-- Performing Test COMPILER_SUPPORTS_STD_CXX11
-- Performing Test COMPILER_SUPPORTS_STD_CXX11 - Success
-- +++ processing catkin package: 'mavros_msgs'
-- ==> add_subdirectory(repo-under-test/mavros_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- mavros_msgs: 25 messages, 27 services
-- +++ processing catkin package: 'mavros'
-- ==> add_subdirectory(repo-under-test/mavros)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0")
-- Configuring done
-- Generating done
-- Build files have been written to: /ros_ws/build
Scanning dependencies of target _mavros_msgs_generate_messages_check_deps_FileRemoveDir
[... 541 lines cut ...]
Generating C++ code from mavros_msgs/FileRemove.srv
[ 78%]
Generating C++ code from mavros_msgs/WaypointClear.srv
[ 78%]
Generating C++ code from mavros_msgs/ParamSet.srv
[ 79%]
Generating C++ code from mavros_msgs/CommandBool.srv
[ 79%]
Generating C++ code from mavros_msgs/WaypointSetCurrent.srv
[ 80%]
[ 80%]
Generating C++ code from mavros_msgs/FileOpen.srv
[ 81%]
Generating C++ code from mavros_msgs/WaypointPull.srv
Generating C++ code from mavros_msgs/CommandTOL.srv
[ 81%]
Generating C++ code from mavros_msgs/WaypointPush.srv
[ 82%]
Generating C++ code from mavros_msgs/FileRemoveDir.srv
[ 82%]
Built target mavros_msgs_generate_messages_cpp
make: *** [all] Error 2

Invoking "make -j8 -l8" failed
```

ChrisTimperley, 2018-11-27:
Issue is avoided by switching from Indigo to Kinetic. Bug is not dependent on a particular ROS version.

closed 2018-11-27 by ChrisTimperley

gavanderhoorn, 2018-11-28:
> Issue is avoided by switching from Indigo to Kinetic. Bug is not dependent on a particular ROS version.

isn't that a contradiction?

If the bug isn't dependent on a particular ROS version, how can switching from one to the other have such an affect as to make the build succeed?

ChrisTimperley, 2018-11-28:
> > Issue is avoided by switching from Indigo to Kinetic. Bug is not dependent on a particular ROS version.
> 
> isn't that a contradiction?
> 
> If the bug isn't dependent on a particular ROS version, how can switching from one to the other have such an affect as to make the build succeed?

Because the Indigo versions of `libmavconn` do not provide support for v2.0 of the `mavlink` protocol. Support for mavlink v2.0, which is required to build the PUT, was added in Kinetic.

gavanderhoorn, 2018-11-29:
So then it is dependent on a particular ROS distribution.

ChrisTimperley, 2018-11-29:
> So then it is dependent on a particular ROS distribution.

The **bug**, described in the `.bug` file (please read: https://github.com/robust-rosin/robust/blob/master/mavros/1a746c3/1a746c3.bug), is not related to a particular version of ROS.

gavanderhoorn, 2018-11-29:
Misunderstood "the bug" here in this case.