robust-rosin/robust #188: Time machine fails for mavros/2998e9f

ChrisTimperley, 2018-11-28:
See: https://github.com/robust-rosin/robust/tree/mavros-2998e9f

```
(robust-q7eP5Wk_) chris@chris-Blade:~/bugs/robust$ python scripts/build-rosinstall.py mavros/2998e9f/
building rosinstall files for directory: mavros/2998e9f/
building rosinstall file for file: mavros/2998e9f/2998e9f.bug
executing command: rosinstall_generator_tm.sh 2014-01-14T02:50:54Z hydro mavros --deps --deps-only --tar
Requested timepoint: '2014-01-14T02:50:54Z' (1389667854)
Switching to pre-REP-141 infrastructure ..
Resetting local rosdistro clone ..
Previous HEAD position was 45edff248... Merge pull request #2866 from stonier/bloom-turtlebot-1
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
Determined rosdistro commit: 45edff24 (authored: 1389664943)
Reverting to ros/rosdistro@45edff24
Cache already exists for (distro; stamp) tuple, skipping generation
Updating local rosdistro index.yaml to use cache from the past ..
Invoking: rosinstall_generator --rosdistro=hydro mavros --deps --deps-only --tar
Using ROS_DISTRO: hydro
The following not released packages/stacks will be ignored: mavros
No packages/stacks left after ignoring not released
time machine failed (return code: 1) for bug [mavros/2998e9f/2998e9f.bug]
```

gavanderhoorn, 2018-11-28:
Looking at [ros/rosdistro@45edff24/hydro/release.yaml](https://github.com/ros/rosdistro/blob/45edff2487ea88acedfd509537a896df529f6b14/hydro/release.yaml), there is no `mavros` there.

So this seems like it is expected behaviour, at least from the time machine.

Because you opened an issue I gather that you didn't expect it though :)

ChrisTimperley, 2018-11-28:
> Looking at [ros/rosdistro@45edff24/hydro/release.yaml](https://github.com/ros/rosdistro/blob/45edff2487ea88acedfd509537a896df529f6b14/hydro/release.yaml), there is no `mavros` there (ctrl+f).
> 
> So this seems like it is expected behaviour, at least from the time machine.
> 
> Because you opened an issue I gather that you didn't expect it though :)

Yikes. That makes this issue even more concerning. Why is `mavros` missing from the `rosdistro`? Looking through the [release history of mavros on GitHub](https://github.com/mavlink/mavros/releases?after=0.7.0), it seems that the first release, `v0.1.0`, was shortly before (5 Jan 2014) the date of the fix commit for this bug (14 Jan 2014).

gavanderhoorn, 2018-11-29:
> Why is `mavros` missing from the `rosdistro`? Looking through the [release history of mavros on GitHub](https://github.com/mavlink/mavros/releases?after=0.7.0), it seems that the first release, `v0.1.0`, was shortly before (5 Jan 2014) the date of the fix commit for this bug (14 Jan 2014).

I don't know exactly yet, but the first version of mavros that appears in `ros/rosdistro` would seem to be `0.5.0` (from [this search](https://github.com/ros/rosdistro/pulls?page=13&q=is%3Apr+mavlink+is%3Aclosed&utf8=%E2%9C%93)).

It could be that earlier versions were released from the precursor to `ros/rosdistro` (that would be an SVN repository).

I'll see if I can find that.

gavanderhoorn, 2018-12-03:
I've not spent too much time on this, but I haven't been able to find any earlier releases @ChrisTimperley. Not in `rosdistro`, neither in something like [ros-dry-releases](http://svn.code.sf.net/p/ros-dry-releases/code/).

At this time I don't know where `0.4.0` and older are.

gavanderhoorn, 2019-05-01:
Unassigned myself as I don't believe this is something I can fix (`mavros` just doesn't seem to have been present in `rosdistro` at the requested time and date).