robust-rosin/robust #213: Error building Kobuki 6e748c1

git-afsantos, 2019-02-13:
The Docker build failed when attempting to build the PUT (`kobuki_node`).

> CMake Error at /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:72 (find_package):
>  Could not find a configuration file for package kdl_conversions.
>
>  Set kdl_conversions_DIR to the directory containing a CMake configuration
>  file for kdl_conversions.  The file will have one of the following names:
>
>    kdl_conversionsConfig.cmake
>   kdl_conversions-config.cmake
>
> Call Stack (most recent call first):
>  repo-under-test/kobuki_auto_docking/CMakeLists.txt:3 (find_package)
>
>
> CMake Error at /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:72 (find_package):
>  Could not find a configuration file for package kobuki_dock_drive.
>
>  Set kobuki_dock_drive_DIR to the directory containing a CMake configuration
>  file for kobuki_dock_drive.  The file will have one of the following names:
>
>    kobuki_dock_driveConfig.cmake
>    kobuki_dock_drive-config.cmake
>
> Call Stack (most recent call first):
>  repo-under-test/kobuki_auto_docking/CMakeLists.txt:3 (find_package)

It seems like some dependencies are missing. Could this be an issue with the timestamp I provided to the time machine? I used the time of the fix, as usual (actually rounded to the next hour, in this case).

The build log is too large to paste here, so let me know if you need it (or need me to look something up within it).

gavanderhoorn, 2019-02-13:
Could as well be another instance of the problem where we're *mysteriously* are missing dependencies even though all should be resolved by `rosdep` (the thing we're trying to address with #150).

gavanderhoorn, 2019-02-13:
> I used the time of the fix, as usual (actually rounded to the next hour, in this case).

Why did you round the stamp?

git-afsantos, 2019-02-13:
@gavanderhoorn no particular reason, just pure laziness, as the rounded time was already in the terminal history, and I figured it would not make much of a difference. But I should probably fix the timestamp, even though it will not solve this particular issue.

gavanderhoorn, 2019-02-13:
Understood.

Retrieving the timestamp should be as simple as `get_issue_creation_date.py https://github/link/to/issue/or/pull_request`.

I know the name of the script only includes `issue`, but it works for PRs as well.

git-afsantos, 2019-03-11:
Just adding a reminder to get back to #258 once this is solved.

ChrisTimperley, 2020-03-12:
This may help to solve the problem? https://github.com/robust-rosin/robust/pull/350