robust-rosin/robust #214: Tested full process of bugzoo container creation

ipa-jfh, 2019-02-18:
Harsh and I today tried the whole robust process (with ac6a181). We can talk about it on Wednesday`s meeting. All in all it worked quite good but still we found some "issues".

### Issues with time machine
```bash
./docker/build.sh # Fails
cd docker && ./build.sh # Works 

```
It is important to be inside the docker folder. One could either fix this or write the steps in the instructions to avoid problems for potential users. 
 
### Issues with bugzoo (@ChrisTimperley )

> Note: This script assumes that the rosinstall_generator_time_machine binary is visible from the PATH.

Maybe more info about  "add  rosinstall_generator_time_machine to path"
(confusing cause I do not see the “binary”?)

---

> $ python scripts/build-rosinstall.py mavros/08cd181.bug (https://github.com/robust-rosin/robust/tree/master/scripts)

should be 08cd181/08cd181.bug 

---

Add in the instructions that one has to create a `test.sh` file.

---

latest bugzoo 2.1.25 is failing.

---

 fork-urls:  without s? (https://github.com/robust-rosin/robust/tree/master/scripts)

---

fork-url cannot handle a list
```bash
Step 5/43 : RUN echo "[ROBUST] cloning repo: '${REPO_FORK_URL}'"  && git clone "${REPO_FORK_URL}" /tmp/repo-under-test  && echo "[ROBUST] cloned repo."
 ---> Running in 5507f593d234
[ROBUST] cloning repo: '['https://github.com/robust-rosin/cob_command_tools']'
Cloning into '/tmp/repo-under-test'...
fatal: I don't handle protocol '['https'
```

I think somewhere earlier though it would complain if it is not a list.

---

Questions:

- how to stop bugzoo container?
- How to delete bugzoo image that has been build?

### Dockerfile

```bash
  File "/home/jfh/stuff_for_ros/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/container.py", line 222, in provision
    raise Exception(msg)  # TODO add exception; DockerException, maybe?
Exception: failed to start Docker container, 7416105e-f718-4cd6-b5fd-1e9ef93322f5:
  [RESPONSE]
      /entrypoint.sh: line 4: /ros_ws/devel/setup.bash: No such file or directory
  [/RESPONSE]
```

This means `catkin_build` failed.

```
Step 42/44 : RUN echo "[ROBUST] attempting to build PUT..."  && echo "[ROBUST] is a build failure expected? ${IS_BUILD_FAILURE}."  && ./build.sh || [ "${IS_BUILD_FAILURE}" = "yes" ]
 ---> Running in acfe3340c7c7
[ROBUST] attempting to build PUT...
[ROBUST] is a build failure expected? yes.
Error(s) in package '/ros_ws/src/repo-under-test/cob_teleop_cob4/package.xml':
The license tag must neither be empty nor only contain whitespaces

Base path: /ros_ws
Source space: /ros_ws/src
Build space: /ros_ws/build
Devel space: /ros_ws/devel
Install space: /ros_ws/install
Creating symlink "/ros_ws/src/CMakeLists.txt" pointing to "/opt/ros/hydro/share/catkin/cmake/toplevel.cmake"
 ---> 6e581e697565
Removing intermediate container acfe3340c7c7
Step 43/44 : COPY test.sh .
 ---> 30569d285e05
```

Build is failing because of missing license in:

https://github.com/ipa320/cob_command_tools/blob/89eb430617039f04c7c61ba0dae8b303c26fc048/cob_teleop_cob4/package.xml

--> How should we solve this?

### TODOs for us (@ipa-jfh  and @ipa-hsd)

- [ ] fix `fix hash` to 
 https://github.com/ipa320/cob_command_tools/commit/81aa175d76608fbac6655ce745e8eaa1d35f51f9#diff-a2b99615ad18e3db8af7eb5c52d6b8af
- [ ] it is not a build bug
- [ ] write test : rosrun cob_script_server cob_console

gavanderhoorn, 2019-02-18:
> ### Issues with time machine
> 
> ```shell
> ./docker/build.sh # Fails
> cd docker && ./build.sh # Works 
> ```
> 
> It is important to be inside the docker folder. One could either fix this or write the steps in the instructions to avoid problems for potential users.

hm, yes. There is definitely the assumption that you run that script from the `docker` directory.

I'll update the docs, but I was sort of expecting users/people to be aware of this :)

gavanderhoorn, 2019-02-20:
@ipa-jfh: I've actually done it somewhat differently: I've made sure the image always gets built from the `docker` sub dir. See https://github.com/rosin-project/rosinstall_generator_time_machine/commit/153b9f6138661e8c1ed70311ad742bf68785244f.

gavanderhoorn, 2019-02-20:
> ### Issues with bugzoo (@ChrisTimperley )
> 
> > Note: This script assumes that the rosinstall_generator_time_machine binary is visible from the PATH.
> 
> Maybe more info about "add rosinstall_generator_time_machine to path"
> (confusing cause I do not see the “binary”?)

Would #210 address this sufficiently?

ipa-jfh, 2019-02-21:
> @ipa-jfh: I've actually done it somewhat differently: I've made sure the image always gets built from the docker sub dir. See rosin-project/rosinstall_generator_time_machine@153b9f6.

I tested it and works :+1: 

> Would #210 address this sufficiently?

This I did not understand. It reads like the script `rosinstall_generator_tm.sh` adds `rosinstall_generator_time_machine` to the path. Or do you mean that `rosinstall_generator_tm.sh` should be visible on the  path?

gavanderhoorn, 2019-02-21:
> > Would #210 address this sufficiently?
> 
> This I did not understand. It reads like the script `rosinstall_generator_tm.sh` adds `rosinstall_generator_time_machine` to the path. Or do you mean that `rosinstall_generator_tm.sh` should be visible on the path?

The sentence that #210 changes reads:

> This script assumes that the `rosinstall_generator_time_machine` binary is visible from the `PATH` (ie: `rosinstall_generator_tm.sh`).

To me it's clear that "the binary" is later clarified by the "ie: `rosinstall_generator_tm.sh`" part, but perhaps it's not.

It's not actually a binary of course, it's a script.

gavanderhoorn, 2019-02-21:
Just submitted #218 which probably clarifies the time machine setup situation better.

ipa-jfh, 2019-02-21:
Ok perfect now :+1:

ChrisTimperley, 2019-02-26:
> fork-urls: without s? (https://github.com/robust-rosin/robust/tree/master/scripts)
> 
> fork-url cannot handle a list

`fork-urls` is correct (example: https://github.com/robust-rosin/robust/blob/master/mavros/1f01916/1f01916.bug). For now, `build-bugzoo.py` script should complain if more than one URL is provided, however. A few infrastructure changes are needed to support multiple PUTs (see https://github.com/robust-rosin/robust/issues/162).

ipa-hsd, 2019-02-27:
@ipa-jfh wrote:
> >fork-urls: without s? (https://github.com/robust-rosin/robust/tree/master/scripts)
fork-url cannot handle a list

@ChrisTimperley wrote:
>`fork-urls` is correct (example: https://github.com/robust-rosin/robust/blob/master/mavros/1f01916/1f01916.bug).

I get the same error while trying to build `care-o-bot/c8091b6` as well.

```
generating manifest for bug file [care-o-bot/c8091b6/c8091b6.bug]
scripts/build-bugzoo.py:40: UserWarning: bad bug file [care-o-bot/c8091b6/c8091b6.bug]: no fork-url
  warnings.warn(m)
```

But the manifest is generated fine when I modify it as: 

`fork-url: https://github.com/robust-rosin/cob_command_tools`

The [build-bugzoo.py](https://github.com/robust-rosin/robust/blob/master/scripts/build-bugzoo.py#L75) in fact looks for `fork-url`and not `fork-urls`. Are we missing something basic here?

ChrisTimperley, 2019-02-27:
> The [build-bugzoo.py](https://github.com/robust-rosin/robust/blob/master/scripts/build-bugzoo.py#L75) in fact looks for `fork-url`and not `fork-urls`. Are we missing something basic here?

No, you're quite right -- my mistake. There's an unmerged PR that switches `fork-url` to `fork-urls`: https://github.com/robust-rosin/robust/pull/168. I'll go ahead and merge that.