robust-rosin/robust #215: building mavros c8091b6

wasowski, 2019-02-19:
Hi @ChrisTimperley , we are trying to build c8091b6 (mavros). Is this something that should be easily doable? Could you check? (possibly after your FSE deadline).

@andersfischernielsen is another student at ITU who is trying to build it.  If you have any hints for him, he is now tagged.

gavanderhoorn, 2019-02-20:
@wasowski: are you referring to the correct bug id?

Afaict, `c8091b6` is not a mavros bug, but a `care-o-bot` one: [robust-rosin/robust/care-o-bot/c8091b6/c8091b6.bug](https://github.com/robust-rosin/robust/blob/a040948df9b7c1b0629cf0f35c6f2a1d1a4b40e2/care-o-bot/c8091b6/c8091b6.bug).

wasowski, 2019-02-20:
@andersfischernielsen see the above answer from @gavanderhoorn: can you confirm the bug number?

andersfischernielsen, 2019-02-23:
Yes, the bug is a `care-o-bot` bug. 👍

gavanderhoorn, 2019-02-24:
Afaik no one has done any work on that.

Perhaps @ipa-jfh knows something more?

ipa-jfh, 2019-02-25:
We have no worked on this one.

@andersfischernielsen for some how-to s have a look at.
- How to use bugzoo: https://github.com/robust-rosin/robust
- How to install the time-machine https://github.com/rosin-project/rosinstall_generator_time_machine
- In order to add the bug to bugzoo this are the steps to follow https://github.com/robust-rosin/robust/tree/master/scripts

They worked quite well for us (with minor issues https://github.com/robust-rosin/robust/issues/214)

ipa-hsd, 2019-02-27:
I built the image for this one: https://github.com/ipa-hsd/robust/tree/bug_c8091b6

I had some issues with it, which I will discuss in a new thread.

wasowski, 2019-03-01:
@andersfischernielsen if you have success please share it here. So that @ipa-jfh can benefit from it.