robust-rosin/robust #262: building care-o-bot 79867c8

ipa-hsd, 2019-03-13:
While building the [bug](https://github.com/robust-rosin/robust/blob/master/care-o-bot/79867c8/79867c8.bug), I get the following error:

```
Updating local rosdistro index.yaml to use cache from the past ..
Invoking: rosinstall_generator --rosdistro=indigo cob_android --deps --tar --deps-only
Using ROS_DISTRO: indigo
The following not released packages/stacks will be ignored: cob_android
No packages/stacks left after ignoring not released
Traceback (most recent call last):
  File "scripts/build-rosinstall.py", line 57, in _time_machine
    stdout=subprocess.PIPE)
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['rosinstall_generator_tm.sh', '2015-08-03T08:03:34Z', 'indigo', 'cob_android', '--deps', '--tar', '--deps-only']' returned non-zero exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scripts/build-rosinstall.py", line 173, in <module>
    main()
  File "scripts/build-rosinstall.py", line 166, in main
    build_file(fn_or_dir, overwrite=args.overwrite)
  File "scripts/build-rosinstall.py", line 118, in build_file
    deps.update(_time_machine(ros_pkgs, dt, distro, deps_only=True))
  File "scripts/build-rosinstall.py", line 61, in _time_machine
    err.returncode, fn_bug_desc)
NameError: name 'fn_bug_desc' is not defined
```
The issue was created at `2015-08-03T08:03:34Z` and the package was first released on [Aug 29, 2015](https://github.com/ipa320/cob_android/releases). So basically, there is no entry of this package for the requested `{TIMEPOINT}` in rosdistro.

How can we solve this?

git-afsantos, 2019-03-13:
I think I ran into a similar issue once or twice before, and I think there is no alternative other than using the release date of the package. But perhaps, someone else has a better solution.

ipa-hsd, 2019-03-15:
The issue was already [fixed](https://github.com/ipa320/cob_android/pull/8) before the 1st release.

gavanderhoorn, 2019-03-17:
Related: https://github.com/rosin-project/rosinstall_generator_time_machine/issues/11.

ipa-hsd, 2019-03-20:
Should I close this, since it is being tracked elsewhere?

gavanderhoorn, 2019-03-20:
Well, the issue of repositories using source builds and reporting bugs in those is tracked in other issues.

But your problem reproducing `79867c8` is not.

What I would recommend is the following: generate a `.rosinstall` file using the first released version (to get all the dependencies in place) and then continue with the commit hash of the unfixed version (ie: the one that was not released).

That should work.

We should make a note about this (ie: not using the "correct" timepoint).

ipa-hsd, 2019-03-20:
I was able to generate `deps.rosinstall` from the following command:
```
rosinstall_generator_tm.sh 2015-08-30T08:03:34Z indigo cob_android --deps --deps-only --tar > deps.rosinstall
```
I guess now I have to modify the `cob_android` package entries manually to `79867c8`.

gavanderhoorn, 2019-03-20:
> I guess now I have to modify the `cob_android` package entries manually to `79867c8`.

yes, if that is the hash for the buggy state, then that would be the idea :)

ipa-hsd, 2019-03-20:
These are the `cob_android` package entries in the `deps.rosinstall`:
```
- tar:
    local-name: cob_android/cob_android_msgs
    uri: https://github.com/ipa320/cob_android-release/archive/release/indigo/cob_android_msgs/0.1.1-0.tar.gz
    version: cob_android-release-release-indigo-cob_android_msgs-0.1.1-0
- tar:
    local-name: cob_android/cob_android_resource_server
    uri: https://github.com/ipa320/cob_android-release/archive/release/indigo/cob_android_resource_server/0.1.1-0.tar.gz
    version: cob_android-release-release-indigo-cob_android_resource_server-0.1.1-0
- tar:
    local-name: cob_android/cob_android_script_server
    uri: https://github.com/ipa320/cob_android-release/archive/release/indigo/cob_android_script_server/0.1.1-0.tar.gz
    version: cob_android-release-release-indigo-cob_android_script_server-0.1.1-0
- tar:
    local-name: cob_android/cob_android_settings
    uri: https://github.com/ipa320/cob_android-release/archive/release/indigo/cob_android_settings/0.1.1-0.tar.gz
    version: cob_android-release-release-indigo-cob_android_settings-0.1.1-0
```

gavanderhoorn, 2019-03-20:
`cob_android` is a metapackage, so cannot actually be the package any bug with `install(..)` statements is in.

Looking at the bug description, it would appear the affected packages are `cob_android_msgs` and `cob_android_script_server` ([here](https://github.com/robust-rosin/robust/blob/c01a6d3f8279996b10b6d99c4aa471bbfcae70bd/care-o-bot/79867c8/79867c8.bug#L54)). Those packages should be passed to the time machine, not the metapackage.

gavanderhoorn, 2019-03-20:
Note: the bug description would seem to be incorrect. The bug is not in `cob_android`, but in those other packages. [This line](https://github.com/robust-rosin/robust/blob/c01a6d3f8279996b10b6d99c4aa471bbfcae70bd/care-o-bot/79867c8/79867c8.bug#L41) should probably be updated.

ipa-hsd, 2019-03-21:
Can someone give me permissions to fork repositories to this organization?

gavanderhoorn, 2019-03-21:
No (as it's not a permission we can give to individual users), but we can invite you to the `robust-rosin` organisation. Which I just did.

ipa-hsd, 2019-03-21:
I don't remember if @ChrisTimperley said if bugzoo can handle multiple PUTs yesterday, but the bug file looks like this:

```
bug:
  package: cob_android_msgs cob_android_script_server
  issue: https://github.com/ipa320/cob_android/issues/7
  time-reported: 2015-08-03 (10:09)
time-machine:
  ros_distro: indigo
  ros_pkgs:
    - cob_android_msgs
    - cob_android_script_server
  issue: https://github.com/ipa320/cob_android/issues/7
bugzoo:
  fork-urls: https://github.com/robust-rosin/cob_android
  bug-commit: 21126f5e3d2e15ae8c93705dc3e8e999707da48e
  fix-commit: 79867c88acc64342f061e6324ed6b84ca7e0773d
```
And the output of 
```
 python build-bugzoo.py
```
is 
```
generating manifest for bug file [care-o-bot/79867c8/79867c8.bug]
build-bugzoo.py:71: UserWarning: BugZoo file does not currently support multiple PUTs
  warnings.warn('BugZoo file does not currently support multiple PUTs')
```

gavanderhoorn, 2019-03-21:
IIRC, BugZoo: not tested (probably not). The `.rosinstall` script: most likely yes.

ChrisTimperley, 2019-03-26:
> ```
> generating manifest for bug file [care-o-bot/79867c8/79867c8.bug]
> build-bugzoo.py:71: UserWarning: BugZoo file does not currently support multiple PUTs
>   warnings.warn('BugZoo file does not currently support multiple PUTs')
> ```

#264 should add multiple PUT support to `build-bugzoo.py.

Also, `fork-urls` should be a list:

```
bugzoo:
  fork-urls:
    - https://github.com/robust-rosin/cob_android
```

ipa-hsd, 2019-03-28:
@ChrisTimperley I am able to build the manifest with https://github.com/robust-rosin/robust/pull/264. Thanks!

Now facing https://github.com/robust-rosin/robust/issues/69

gavanderhoorn, 2019-03-28:
> Now facing #69

#69? Can you show the build log for your bug?

Has the `care-o-bot` repository been renamed? Renames of `geometry_experimental` should be automatically handled.

Could it be that you're running into vcstools/vcstools#147?

ipa-hsd, 2019-03-28:
@gavanderhoorn : Sorry that was a hasty reply as I didn't dig into the error.

I am getting similar errors for `tf2` related packages:
```
ERROR [vcstools] Tarball download unpack failed: u"geometry_experimental-release-release-indigo-tf2-0.5.12-0 is not a subdirectory with contents in members [u'geometry2-release-release-indigo-tf2-0.5.12-0']"[/vcstools]
```
```
ERROR in config: Error processing 'geometry_experimental/tf2' : [geometry_experimental/tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.12-0.tar.gz version geometry_experimental-release-release-indigo-tf2-0.5.12-0 into /ros_ws/src/geometry_experimental/tf2 failed.
```

gavanderhoorn, 2019-03-28:
That would seem to be #69, but afaik we're already handling that with automatic renames of entries in `.rosinstall` files.

Can you clarify which versions of things you are using, and which commands lead to this error?

ipa-hsd, 2019-03-28:
I pulled https://github.com/robust-rosin/robust/pull/264 in a new branch and working on it.

These are the tars in deps.rosinstall which fail:
```
- tar:
    local-name: geometry_experimental/tf2
    uri: https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2/0.5.12-0.tar.gz
    version: geometry_experimental-release-release-indigo-tf2-0.5.12-0
- tar:
    local-name: geometry_experimental/tf2_msgs
    uri: https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_msgs/0.5.12-0.tar.gz
    version: geometry_experimental-release-release-indigo-tf2_msgs-0.5.12-0
- tar:
    local-name: geometry_experimental/tf2_py
    uri: https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_py/0.5.12-0.tar.gz
    version: geometry_experimental-release-release-indigo-tf2_py-0.5.12-0
- tar:
    local-name: geometry_experimental/tf2_ros
    uri: https://github.com/ros-gbp/geometry_experimental-release/archive/release/indigo/tf2_ros/0.5.12-0.tar.gz
    version: geometry_experimental-release-release-indigo-tf2_ros-0.5.12-0
```

The following command lead to the error:
```
bugzoo bug build --force robust:79867c8
```

gavanderhoorn, 2019-03-28:
Renaming happens when the `.rosinstall` file is generated ([here](https://github.com/robust-rosin/robust/blob/db9bfadff62275e6f7a7f7a0a9b91974fa6f8d07/scripts/build-rosinstall.py#L129-L135)), but only if you actually used `scripts/build-rosinstall.py`.

Did you run the time machine manually?

ipa-hsd, 2019-03-28:
> Ah, just saw that you did: #262 (comment).

Was about to comment that :) 

Can I use `scripts/build-rosinstall.py` when I am selecting the {TIMEPOINT} manually?

gavanderhoorn, 2019-03-28:
> Can I use `scripts/build-rosinstall.py` when I am selecting the {TIMEPOINT} manually?

Well .. no, not really (if I understand you correctly).

`scripts/build-rosinstall.py` will use the `time-machine/timestamp` key in the `.bug` description to drive the time machine. In this case I would just manually replace `geometry_experimental` with `geometry2` and commit the updated `.rosinstall`.

Note that you may still run into vcstools/vcstools#147 unfortunately.

ipa-hsd, 2019-03-28:
Manually replacing `geometry_experimental` with `geometry2` helps.

But docker build in 
```
bugzoo bug build --force robust:79867c8
```
still fails with 
```
/ros_ws/src/cob_driver/cob_sound/ros/src/sound.cpp:9:32: fatal error: cob_srvs/SetString.h: No such file or directory
 #include <cob_srvs/SetString.h>
                                ^
compilation terminated.

make[2]: *** [CMakeFiles/sound.dir/ros/src/sound.cpp.o] Error 1

make[1]: *** [CMakeFiles/sound.dir/all] Error 2

make: *** [all] Error 2

Command failed, exiting.
```
Looks like [cob_sound-0.6.4](https://github.com/ipa320/cob_driver/tree/0.6.4) (which is currently listed in `deps.rosintall`) did not include [cob_common](https://github.com/ipa320/cob_common) as its dependency. It was done in [cob_sound-0.6.5](https://github.com/ipa320/cob_driver/releases/tag/0.6.5)

gavanderhoorn, 2019-03-28:
That would be a candidate for `missing-dependencies` then. See #150.

ipa-hsd, 2019-03-28:
But don't I have to use `scripts/build-rosinstall.py` for `missing-dependencies` as well?
I might have to manually change the release version in this case.

gavanderhoorn, 2019-03-28:
> But don't I have to use `scripts/build-rosinstall.py` for `missing-dependencies` as well?

If you want to have everything done automatically: yes.

But in the end all it does is run the time machine twice. Once for the actual PUT(s) (and with `--deps-only`) and once for the missing dependencies (without `--deps-only`). See also #150.

> I might have to manually change the release version in this case.

Don't believe that's needed in this case.

As this is a special case (bug in unreleased pkg) I'd just run the time machine manually, as you're already doing.

ipa-hsd, 2019-03-28:
in `scripts/build-rosinstall.py`
```
        deps.update(_time_machine(ros_pkgs, dt, distro, deps_only=True))
        if missing_deps:
            deps.update(_time_machine(missing_deps, dt, distro, deps_only=False))
```
Makes sense.

Ran the command
```
rosinstall_generator_tm.sh 2015-09-01T08:03:34Z indigo cob_common --deps --tar > deps.rosinstall
```
`deps.rosintall` looks file now. 

But https://github.com/robust-rosin/robust/pull/264#issuecomment-477648509

ipa-hsd, 2019-04-03:
Guess I was unclear, but I am able to build the image now.

gavanderhoorn, 2019-04-03:
@ipa-hsd: with #270 merged, you should be able to set the timestamp you used for the `.rosinstall` files (ie: the one after the first release) in the `.bug` file, and Bugzoo should make the time machine use that instead of the timestamp of the issue (which would be too early).

It would be great if you could test whether that actually works.

ipa-hsd, 2019-04-03:
I merged fetched the `master` (which should include https://github.com/robust-rosin/robust/pull/270) branch and rebased https://github.com/robust-rosin/robust/pull/264

I am able to build the image normally now.

gavanderhoorn, 2019-04-03:
Which timestamp did you configure for the time machine to use?

ipa-hsd, 2019-04-03:
`2015-08-30T08:03:34Z`, which is 1 date after the 1st release of [`cob_android`](https://github.com/ipa320/cob_android/releases)

closed 2019-04-04 by ipa-hsd

gavanderhoorn, 2019-04-04:
@ipa-hsd: perhaps you've already done so, but could I ask you to add a comment to the `.bug` file explaining that timestamp (ie: that it's a datetime later than the original reporting time of the issue, due to the package not being released, etc)?

ipa-hsd, 2019-04-04:
I hadn't, will do so :+1: