robust-rosin/robust #269: build-rosinstall: should it use datetime before using issue url?

gavanderhoorn, 2019-03-28:
`scripts/build-rosinstall.py` currently tries to retrieve the timestamp of a bug report using the Github API if it finds an `issue` key in the `time-machine` section:

https://github.com/robust-rosin/robust/blob/784b6f6e619a3e1588a9b47521299dc82c2bad51/scripts/build-rosinstall.py#L101-L112

If it doesn't find one, it tries to use a `datetime` key in the same section.

I'm wondering whether this should be reversed: use `datetime` if available, if not, try using the `issue` URL with the GH API.

Rationale: we might want to be able to specify a different timestamp for the time machine to use. Different from whatever timestamp GH returns when asking it for the stamp of a particular issue or PR. This is especially true in cases where we're replicating bugs in unreleased packages.

gavanderhoorn, 2019-03-28:
Example of a bug where reversing this would help: #262.

The `datetime` key in the `time-machine` section could be commented to explain why a different timestamp is being specced.

gavanderhoorn, 2019-03-28:
@ChrisTimperley

ChrisTimperley, 2019-04-01:
I agree. I'll go ahead and tweak the script.

commit 8e2a1b29 (2019-04-01): Swapped order of datetime and issue checks in build-rosinstall (fixes #269) (#270)
closed 2019-04-01 by ChrisTimperley