robust-rosin/robust #275: ur_kinematics not released

git-afsantos, 2019-05-07:
I picked up `c779432` to reproduce, but running the time machine with the pull request creation date says the package had not been released yet.

I ran `git blame` on `rosdistro` to find out the release date in Kinetic and used that to run the time machine again. Still no luck. I even tried a date that is one hour later than the release, and no luck either.

`git blame` gives me

> 7ee6cbd0 (ipa-fxm 2017-08-04 10:10:41 +0200 15010)

which is already several months after the issue date, but the time machine says

> Reverting to ros/rosdistro@7afbecd1

Which seems to have been pushed some time later, but for some reason (branching issues?) does not contain `ur_kinematics`.

@gavanderhoorn Any suggestion to find a suitable date?

gavanderhoorn, 2019-05-08:
Could you please post the complete output of the time machine?

gavanderhoorn, 2019-05-08:
`universal_robot` was indeed only released into Kinetic on the 4th of August (ros/rosdistro#15689). Before that I would expect the time machine to complain (it's really `rosinstall_generator` btw that complains).

How are you running the time-machine? Still manually, or using the script that @ChrisTimperley wrote?

Can you provide some details on your workflow and commands used?