robust-rosin/robust #279: Bugzoo fails to build c779432

git-afsantos, 2019-05-08:
Bugzoo data:

```yaml
bugzoo:
  fork-urls: 
    - https://github.com/robust-rosin/universal_robot
  is-build-failure: yes
  use-osrf: no  # [optional]
  bug-commit: 7958c81afed1d70322efbd84aa042ddfaf2c54ea
  fix-commit: c7794329bf632592bf1d345d1290ff4b8d13580a
```

Command:

```bash
bugzoo bug build --force robust:c779432 2>&1 | tee universal_robot/c779432/build.log
```

Error:

> E: Unable to locate package libfcl-0.5-dev
> E: Couldn't find any package by glob 'libfcl-0.5-dev'
> E: Couldn't find any package by regex 'libfcl-0.5-dev'
> ERROR: the following rosdeps failed to install
>   apt: command [apt-get install -y libfcl-0.5-dev] failed

Log (will remove it from the repository later):
https://github.com/robust-rosin/robust/blob/ec9906a79e954fa42179ba8779eb0fe4508df2de/universal_robot/c779432/build.log

git-afsantos, 2019-05-08:
A quick search leads me to believe that switching to `use-osrf: yes` might be necessary. Will try it out.

gavanderhoorn, 2019-05-08:
What is the targeted ROS distro here?

git-afsantos, 2019-05-08:
ROS Kinetic, for this particular bug.

gavanderhoorn, 2019-05-08:
For Kinetic that package indeed is provided through the OSRF package repository.

gavanderhoorn, 2019-05-08:
Is this package needed because of MoveIt?

If so: we might want to see whether that is really needed. Building MoveIt from source is costly, and should only be needed in very few circumstances.

Do you already have the updated `.bug` file pushed somewhere?

git-afsantos, 2019-05-08:
Here is the updated `.bug` file:
https://github.com/robust-rosin/robust/blob/reproduce-c779432/universal_robot/c779432/c779432.bug

gavanderhoorn, 2019-05-08:
Hm, because this bug is specifically about `ur_kinematics`, we can't really help that: compiling that unfortunately depends on MoveIt.

git-afsantos, 2019-05-08:
After a few hours the build completed and everything works as expected.

closed 2019-05-08 by git-afsantos

gavanderhoorn, 2019-05-08:
What is the size of the image?

git-afsantos, 2019-05-08:
`docker image inspect` reports `"Size": 2712778611` (in Bytes). I assume this value takes into account all the layers.