robust-rosin/robust #292: Error building `care-o-bot/ca7b536`

ipa-hsd, 2019-05-16:
While running the `rosinstall-generator` with the following command: 

```
python scripts/build-rosinstall.py care-o-bot/ca7b536/ca7b536.bug
```
I get the error that `cob_obstacle_distance` is not released at the time the bug was reported:
```
building rosinstall file for file: care-o-bot/ca7b536/ca7b536.bug
overwriting file: care-o-bot/ca7b536/deps.rosinstall
executing command: rosinstall_generator_tm.sh 2015-08-03T12:15:20Z indigo cob_obstacle_distance --deps --tar --deps-only
Requested timepoint: '2015-08-03T12:15:20Z' (1438604120)
Resetting local rosdistro clone ..
Previous HEAD position was 311af91... Merge pull request #9291 from Karsten1987/bloom-naoqi_bridge_msgs-0
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
Determined rosdistro commit: 3546c974 (authored: 1438606361)
Reverting to ros/rosdistro@3546c974
Cache already exists for (distro; stamp) tuple, skipping generation
Updating local rosdistro index.yaml to use cache from the past ..
Invoking: rosinstall_generator --rosdistro=indigo cob_obstacle_distance --deps --tar --deps-only
Using ROS_DISTRO: indigo
The following not released packages/stacks will be ignored: cob_obstacle_distance
No packages/stacks left after ignoring not released
time machine failed (return code: 1)
Traceback (most recent call last):
  File "scripts/build-rosinstall.py", line 192, in <module>
    main()
  File "scripts/build-rosinstall.py", line 185, in main
    build_file(fn_or_dir, overwrite=args.overwrite)
  File "scripts/build-rosinstall.py", line 119, in build_file
    deps.update(_time_machine(ros_pkgs, dt, distro, deps_only=True))
TypeError: 'NoneType' object is not iterable
```
So I added the `datetime` tag to `time-machine`:
```
time-machine:
  ros_distro: indigo
  ros_pkgs:
    - cob_obstacle_distance
  issue: https://github.com/ipa320/cob_control/issues/48
  datetime: 2015-08-26
```
which is 1 day after the package was released. After that I was able to generate the `deps.rosinstall` file.

Now I notice that in `robust.bugzoo.yml`, there are 2 images for each bug: `:bug` and `:fix`. Guess it is from here: https://github.com/robust-rosin/robust/commit/ecc6facfb816b91979904e377f7e9e19c41a2e96

Do I build these 2 images separately? What happens to the ones I have already built for previous bugs, should I rebuild them all?

Also while building the bugzoo images:
```
bugzoo bug build --force robust:ca7b536:bug
``` 
I get the following error:
```
 ---> Running in a92f907b0043
[ROBUST] fetching latest buggy and fixed verisons...
[ROBUST] fetching bug version: f596e45f4fd43ea2d696e32c966314e107aee84a
[ROBUST] fetching fix version: ca7b536390bcf76bf4ab6e62354c52bd51fb1ece
Fetching origin
[ROBUST] fetched latest buggy and fixed versions.
[ROBUST] generating patch diff...
fatal: bad object ca7b536390bcf76bf4ab6e62354c52bd51fb1ece

Traceback (most recent call last):
  File "/home/hsd/bugzoo_venv/bin/bugzoo", line 10, in <module>
    sys.exit(main())
  File "/home/hsd/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 541, in main
    args.func(args)
  File "/home/hsd/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 489, in <lambda>
    cmd.set_defaults(func=lambda args: build_bug(rbox, args.bug, args.force))
  File "/home/hsd/bugzoo_venv/lib/python3.5/site-packages/bugzoo/cli/__init__.py", line 103, in build_bug
    rbox.bugs.build(bug, force=force)
  File "/home/hsd/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/bug.py", line 82, in build
    quiet=quiet)
  File "/home/hsd/bugzoo_venv/lib/python3.5/site-packages/bugzoo/mgr/build.py", line 124, in build
    raise ImageBuildFailed(name, log)
bugzoo.exceptions.ImageBuildFailed: failed to build Docker image: robustrosin/robust:ca7b536-fix
```

gavanderhoorn, 2019-05-29:
Looks like `git` can't find commit `ca7b5363` in the clone of `robust-rosin/cob_control`.

It does exist though: https://github.com/robust-rosin/cob_control/commit/ca7b536390bcf76bf4ab6e62354c52bd51fb1ece.

gavanderhoorn, 2019-05-29:
@ChrisTimperley: would you have any idea? Could this be related to this: https://github.com/robust-rosin/robust/blob/9c5e6997378b8f83e2758abd80acb265ece3dabd/Dockerfile#L39-L54

ipa-hsd, 2019-06-03:
Done.

https://github.com/robust-rosin/robust/tree/care-o-bot/ca7b536

ChrisTimperley, 2019-06-03:
Interesting. I'm able to reproduce your build failure via BugZoo, and by running the same Git commands natively on my machine:
```
chris@nostromo:~$ cd /tmp/
chris@nostromo:/tmp$ git clone https://github.com/robust-rosin/cob_control
Cloning into 'cob_control'...
remote: Enumerating objects: 13805, done.
remote: Total 13805 (delta 0), reused 0 (delta 0), pack-reused 13805
Receiving objects: 100% (13805/13805), 5.71 MiB | 11.37 MiB/s, done.
Resolving deltas: 100% (8849/8849), done.
chris@nostromo:/tmp$ cd cob_control/
chris@nostromo:/tmp/cob_control$ git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
chris@nostromo:/tmp/cob_control$ git fetch --all
Fetching origin
chris@nostromo:/tmp/cob_control$ git diff f596e45f4fd43ea2d696e32c966314e107aee84a ca7b536390bcf76bf4ab6e62354c52bd51fb1ece
fatal: bad object ca7b536390bcf76bf4ab6e62354c52bd51fb1ece
```

ChrisTimperley, 2019-06-03:
Looking at the associated PR, there's a lot of force pushing going on: https://github.com/ipa320/cob_control/pull/47

gavanderhoorn, 2019-06-03:
Yes. Indeed.

Wouldn't https://github.com/ipa320/cob_control/commit/9dac57a1a6f4c9edddd2125c5f8b9409526a26bf be the correct commit for `fix`?

gavanderhoorn, 2019-06-03:
Or otherwise https://github.com/ipa320/cob_control/pull/47/commits/47adcee14c7f39645352f2226b32c5f116ea0c22?

ipa-hsd, 2019-06-07:
Thanks for your inputs on Wednesday. I am able to build the image with  ipa320/cob_control@47adcee

closed 2019-06-07 by ipa-hsd