robust-rosin/robust #355: YAML schema checker: yamale

gavanderhoorn, 2020-08-13:
Would [yamale](https://github.com/23andMe/Yamale) perhaps be an option to use, instead of the current yaml->json route?

gavanderhoorn, 2020-08-13:
@wasowski @ipa-hsd ?

ipa-hsd, 2020-08-13:
Will try this out. Thanks @gavanderhoorn

gavanderhoorn, 2020-08-26:
Ping?

Would probably require different/migrated schema, but seems more efficient as it avoids the conversion.

ipa-hsd, 2020-08-26:
It looks like the effort out-weighs the benefits (conversion is just another command in the script or it has other "problems"?). For now I would focus on cleaning up the bug files.

gavanderhoorn, 2020-08-26:
IIUC, yamale will also give you line nrs or direct references to where schema violations are located.

Wasn't that one of the problems you had with the current approach?

I also expect yamale to be much faster than the current approach.

ipa-hsd, 2020-08-26:
> yamale will also give you line nrs or direct references to where schema violations are located.

My main problem is that it exits after the 1st error it encounters, so I don't get an overview of errors. But I could make a small test for only a section of the bug file to see if I can automate the changes I am making.

gavanderhoorn, 2020-08-26:
Don't just do it for me of course.

I just wanted to let 'us' know about a possible alternative tool.

ipa-hsd, 2020-08-26:
Noted :+1:

gavanderhoorn, 2021-02-10:
I believe yamale is now being used.

so closing

closed 2021-02-10 by gavanderhoorn