robust-rosin/robust #357: Include robot specific languages in schema?

ipa-hsd, 2020-08-26:
The [taxonomy description](https://github.com/robust-rosin/robust/blob/60c2902d5069f015027231b4a51096b7b900bcfa/doc/taxonomy-description.txt#L91) assumes that the bugs are originating in ROS-specific files. [This](https://github.com/robust-rosin/robust/blob/3aed0b3de965c50ba3d6b58142c2f20ebbe41500/universal_robot/cda133d/cda133d.bug#L35) bug file for example has a bug originating in `URScript`. 

Should we add robot-specific languages in the schema as well? Not sure if there are more examples related to ABB and other robots as well?

ipa-hsd, 2020-08-26:
@ChrisTimperley @gavanderhoorn @wasowski ?

gavanderhoorn, 2020-08-26:
Perhaps adding an `other` category might be an idea.

There are very few bugs which would fall in that category, so not sure whether it would be worth it to add RAPID and URScript.

gavanderhoorn, 2020-08-26:
If it aren't that many languages, we could also just add them to the schema.

@ipa-hsd: would RAPID and URscript be the only two? Or are there Motoman bugs which were reported to be in the INFORM jobs?

ipa-hsd, 2020-08-27:
The only reference I found to `INFORM` was in [here](https://github.com/robust-rosin/robust/blob/master/motoman/377d7be/377d7be.bug), though it is not language specific bug. But I agree with your previous comment, else the schema will have to be expanded each time a new robot is added (not in case of this study of course). Just that instead of `other`, I suggest something like `robot specific`.