robust-rosin/robust #358: Add "ignore file" marker to pre 2013 bugs?

gavanderhoorn, 2020-08-26:
As discussed in our robust meeting (2020-08-26): a nr of bugs (mostly in `kobuki`) are older than the 22nd of April, 2013 and the time-machine cannot go back that far.

Right now those bugs are stored in a separate directory, but this breaks the links we're using in the paper.

Suggestion: add *some way* in which we can mark those bugs as "too old", so we can avoid building images for them, without having to store those bug descriptions in a different directory.

This could be something we add to BugZoo, the bug schema or the time-machine.

The latter doesn't really read bug description files though, it's just a wrapper around `rosinstall_generator`, and it already errors out when trying to go back too far.

@ChrisTimperley: would adding this to BugZoo make sense?

gavanderhoorn, 2020-09-09:
We'll go for the ignore files.

wasowski, 2020-09-09:
for the record: we agreed on ignore files on the telco. The ball is with you @gavanderhoorn  :)

ChrisTimperley, 2020-09-14:
I believe that this has been resolved by #360. @gavanderhoorn @wasowski Feel free to re-open the issue if I'm mistaken!

closed 2020-09-14 by ChrisTimperley