robust-rosin/robust #361: Fix "bug.phase" property in bug reports

ChrisTimperley, 2020-09-16:
In RQ2.1 for the paper, we report the phase in which the associated failure occurs for each bug. That information is currently captured by the `bug.phase` property in the bug reports. Currently, we have a number of different (and redundant) values for this property, shown below.

```
chris@chris-MS-7B77:~/bugs/robust$ grep -h "phase: " -r . | cut -d':' -f2 | sort | uniq
 build-time
 compile-time
 deployment-time
 null
 programming-time
 run-time
 runtime
 runtime-initialization
 runtime-operation
```

I propose that we update the set of acceptable values for the `bug.phase` property to the following:
* `"build"`
* `"deployment"`
* `"runtime"`
* `"none"`

There's some debate to be had about the last value, which we should probably discuss during the meeting today. The idea of having an explicit `"none"` rather than relying on an implicit `null` is to make clear the distinction between the case where we know the associated phase and the case where that information is missing.

wasowski, 2020-09-16:
So my counter proposal (Also for other properties):

- "build"
- "deployment" (e.g., launch failures)
-  "runtime"
- "does not apply"
- "unknown"

The last two case make the reasons for lack of an actual value clear. @gavanderhoorn

gavanderhoorn, 2020-09-23:
We decided to rename `deployment` to better reflect the configuration or startup phase.

commit dd033632 (2020-09-29): Fixed bug.phase in bug reports to be consistent (fixes #361) (#363)

* used does not apply

* changed compile-time to build

* build-time -> build

* handled deployment-time bugs

* run-time -> runtime

* runtime-init -> startup

* fixed runtime issues

* revised to runtime
closed 2020-09-29 by ChrisTimperley