robust-rosin/robust #364: Fix "bug.subsystem" property in bug reports

ChrisTimperley, 2020-09-29:
In RQ2.2 for the paper, we report the subsystems affected by the failure for each bug. That information is currently captured by the bug.subsystem property in the bug reports. We currently have a lot of different values for this property, shown below, that lack cohesion.

```
chris@chris-MS-7B77:~/bugs/robust$ grep -h "subsystem: " -r . | cut -d':' -f2 | sort | uniq
 application-specific component
 controller
 core component
 driver
 "driver | controller # (component getting joystick motion and deducing the desired 6D motion of the arm end effector)"
 generic component
 generic task component
 launch file
 MAVCONN library
 metapackage
 motion
 motion
 motoplus pluging for the controller
 N/A
 null
 orchestration
 package metadata
 robot description
 ROS
 ROS-comm
 ROS tool
 safety controller
 scripts
 simulation plugin
 simulation plugins
 specific application component
 test suite
 visualization
```

We should figure out a small set of reasonable `subsystem` labels. Any suggestions, @gavanderhoorn, @wasowski, @git-afsantos, @ipa-hsd?

ChrisTimperley, 2020-09-30:
Following the meeting, we've decided to drop this RQ from the paper for now since it's difficult to draw any meaningful conclusions from the data.

ipa-hsd, 2020-10-07:
@ChrisTimperley Thanks for the `grep` command! :D