robust-rosin/robust #370: Need a clean figures/e964bbb.bug (kobuki)

wasowski, 2020-10-08:
We need a clean version of YAML file with all the fields final for of kobuki:e964bdd.bug, including the fields for codes.  See the file (how we use it): https://github.com/ChrisTimperley/ese-robust/blob/master/figures/e964bbb.bug (especially field names `failure-codes` and `fault-codes`).

Ping us when you have it, @ipa-hsd. Then we will abbreviate it and format for the paper repo.

ipa-hsd, 2020-10-13:
https://github.com/robust-rosin/robust/pull/359/commits/3726e5b403baa85beec4ee2c00133bedf49e32a5

wasowski, 2020-10-13:
oki.  I will wait with moving it to the paper until our meeting tomorrow.

ipa-hsd, 2020-10-15:
https://github.com/robust-rosin/robust/pull/359/commits/e715bf38a2bca66c2dfc70e98dfa17f106f724d0 changes task to `locomotion` as per yesterday's discussion.

wasowski, 2020-10-28:
@ipa-hsd we still do not have fault-codes and failure-codes for this one (which is needed in the main example in the paper).

wasowski, 2020-10-28:
(mis-clicked close, so re-openning)

ipa-hsd, 2020-10-28:
> 3726e5b. Please note that I have changed the task from differential drive to motion in the bug file and also added ìt to the schema. We can discuss about the taxonomy in our next meeting.

I added the fault and failure codes here which now part of the [master](https://github.com/robust-rosin/robust/blob/master/kobuki/e964bbb/e964bbb.bug) branch. Did I miss-understand something here?

wasowski, 2020-10-28:
Oh - my bad. I looked at the copy I made when you responded to the issue originally.  I missed the edits made later.  I will have a look and com back to you. Thanks!

wasowski, 2020-10-28:
Looks good, @ipa-hsd !

closed 2020-10-28 by wasowski