robust-rosin/robust #379: A small number of bugs involve multiple fix commits in one or more repositories

ChrisTimperley, 2020-10-21:
For example, `0027b57` contains two fix commits in the same repository, and `fd6b589` contains two commits across two repositories. The bug description format should be adjusted to better support such cases. A simple fix is to transform the `repo` and `hash` fields in `fix` into lists. However, this introduces some ambiguity into which commits belong to which repo.

A better alternative is to replace both of these fields with a `commits` field, which describes each of the fix commits. Fix commits would be described by an object specifying `repo` and `hash`, and could also specify some useful information for the purpose of analysis (e.g., `stats`, `files`, `committer`, `date`, etc).

gavanderhoorn, 2020-10-21:
Isn't this a duplicate of #162?

wasowski, 2020-10-21:
go ahead @ChrisTimperley !

commit 12ec1540 (2020-11-18): Updated bug description files to support multiple fix commits (fixes #379) (#382)

* automatically refactored the majority of files

* fixed 928306b

* fixed 9299530

* 3ea2c30

* fixed confidential bugs

* fixed care-o-bot

* added note to b1b6fcb

* fixed motoman bugs

* added note to 045ab54

* fixed kobuki bugs

* fixed geometry2 bug

* fixed turtlebot bug

* fixed confidential bug

* added note to motoman bug

* b1b6fcb: correct fix.hash and fix.pull-request

* 045ab54: correct fix.hash and fix.pull-request

* fixed commits in gijs's forks to upstream commits
closed 2020-11-18 by ChrisTimperley