robust-rosin/robust #389: Average size of Docker images (or a range)?

gavanderhoorn, 2020-11-11:
In order to gauge storage requirements for the Docker images, I'm trying to come up with an average size which can be used to calculate a total storage volume needed to persist them.

It's more likely we have a range. That would also be OK.

gavanderhoorn, 2020-11-11:
@ChrisTimperley: would you have a quick figure here?

gavanderhoorn, 2020-11-11:
I'm also trying to figure out whether it would be possible to store *diffs* between Docker images.

Exporting an image to a `.tar` will make it impossible to exploit any caches or deduplication, which may cause storage requirements to become significantly higher.

Compression will help, but it's unclear how much.

ChrisTimperley, 2020-11-11:
I believe that it's possible to export multiple images to a single tar: https://stackoverflow.com/questions/35575674/how-to-save-all-docker-images-and-copy-to-another-machine