robust-rosin/robust #401: How to run the jupyter notebook?

wasowski, 2021-02-09:
I believe that I followed the instructions fine (installed virtual env, installed requirements, started notebook) but I am getting a crash on importing robust. 

```
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-81fea31c5e96> in <module>
      4 
      5 import IPython
----> 6 import robust
      7 import tabulate
      8 import yaml

ModuleNotFoundError: No module named 'robust'
```

(I wanted to count how many bugs have proper CVE classifications.)

ipa-hsd, 2021-02-10:
Do you think this script might be useful? https://github.com/robust-rosin/robust/pull/400/commits/bcabd4c8f032bdc5c97242ddc2e573a858254965. If the CVE classifications can be added to the schema (`robust.yaml`), it would be easier.

ipa-hsd, 2021-02-10:
did you try
```
pip(3) install <path/robust/repo>
```
?

wasowski, 2021-02-10:
I did  not. Should I add this to instructions in README.rst? Is there no way to include this in the requirements file?

wasowski, 2021-03-03:
fixed in #402

closed 2021-03-03 by wasowski