robust-rosin/robust #407: System failure code for kobuki/9de9690

git-afsantos, 2021-04-07:
[**Report File**](https://github.com/robust-rosin/robust/blob/cdd9f43f576ee1ee6ebe03418666a109ffd098b9/kobuki/9de9690/9de9690.bug)

> Kobuki has a package, `kobuki_ftdi` which contains utilities for flashing and enabling Kobuki's USB connection.
> One such utility copies a udev rule to enable the device to appear on `/dev/kobuki`.
> When users install Turtlebot and Kobuki packages on their system, they often do so with the deb packages for Ubuntu.
> However, `kobuki_ftdi` was not included in the main package, and when the users got to the point where they would set up the udev rule, the process would fail, since `kobuki_ftdi` was not found.
> This package is clearly not a compile time nor runtime dependency, however the solution was to declare it as both, so that users would have to install the package anyway.

```yaml
failure-codes:
  - SOFTWARE:BUILD
  - SYSTEM:LOSS-OF-CONTROL
```

I was under the assumption that all `SOFTWARE:BUILD` failures were, by definition, labelled as `SYSTEM:NONE`, since failures at the build/install stage would never get to the point where the system is booted and expected to do things.
Thus, the *system* would never fail.

Is there a reason or argument for this bug to be considered a loss of control?
Is it because the installation of the udev rule was an *"optional"* (i.e., not automatic or enforced) step?

I would understand that. If users did not follow all the installation instructions, there would be **no** software failure, but they would be unable to control the robot (manually or automatically; no ROS node would get any commands to the robot).

The [reported issue](https://github.com/yujinrobot/kobuki/issues/305) however, complains about a problem when users *are* setting the udev rule, meaning that they are following installations instructions and encountering a bug at *that* stage - not when the system is running.

Should both labels be kept, because the fault can manifest in these two different scenarios, or should we label according with the original report only?

ipa-hsd, 2021-04-14:
> Is it because the installation of the udev rule was an "optional" (i.e., not automatic or enforced) step?

My bad. I reasoned that since it was not a compile time error, the users might still be able to start the system.

>  should we label according with the original report only?

Agreed.

ipa-hsd, 2021-04-28:
Fixed in https://github.com/robust-rosin/robust/pull/399

closed 2021-04-28 by ipa-hsd