robust-rosin/robust #408: Harmless flag not included in bug reports

git-afsantos, 2021-04-07:
Bug reports do not have the `HARMLESS` failure *"flag"*. Did we agree on dropping it, or is it an oversight?

ipa-hsd, 2021-04-14:
Failure codes are coming from table 5 and 6 and not from the ROBUST codebook anymore. `HARMLESS` and `MISLEADING` does not seem to be part of either.

ipa-hsd, 2021-04-14:
Currently the bugs which have been marked `HARMLESS`
```
universal_robot/fa64ec6/fa64ec6.bug
motoman/eb99671/eb99671.bug
kobuki/8163705/8163705.bug
kobuki/e34428d/e34428d.bug
geometry2/b4dc23c/b4dc23c.bug
```
TODO: 
- re-look at bugs which have been flagged `SYSTEM:NONE`
- If above bugs have flag other than `SYSTEM:NONE`, add `HARMLESS`

ChrisTimperley, 2021-04-14:
Following our discussion in the meeting, we've decided that all of the bugs above are `SYSTEM:NONE` and there is no need for the `HARMLESS` tag.

closed 2021-05-05 by ipa-hsd