robust-rosin/robust #410: SYSTEM:MODEL-REALITY-DIVERGENCE or SYSTEM:UNINTENDED-MOVEMENT for TF related issues?

ipa-hsd, 2021-05-03:
[geometry2/860b866/860b866.bug](https://github.com/robust-rosin/robust/blob/master/geometry2/860b866/860b866.bug)
[motoman/1ec8ca1/1ec8ca1.bug](https://github.com/robust-rosin/robust/blob/master/motoman/1ec8ca1/1ec8ca1.bug)
[other/22e4e4f/22e4e4f.bug](https://github.com/robust-rosin/robust/blob/master/other/22e4e4f/22e4e4f.bug)
[other/473ab5e/473ab5e.bug](https://github.com/robust-rosin/robust/blob/master/other/473ab5e/473ab5e.bug)

Though the issues state about "wrong visualization" or "wrong position", `SYSTEM:UNINTENDED-MOVEMENT` would be an immediate consequence. And sometimes (as in case of `geometry2/860b866/860b866.bug`), it depends on the context how the function was used. How strict do we want to be about extrapolation in TF related issues?  

Not sure if we have already discussed these scenarios.

ipa-hsd, 2021-05-05:
Till now we have been using labels that are directly addressed in the description and avoiding inferences. But `other/473ab5e/473ab5e.bug` and even `other/22e4e4f/22e4e4f.bug` seem too close to `UNINTENDED-MOVEMENT` to be left out. Even `geometry2/860b866/860b866.bug` depends on the context (which is not mentioned here). The function `do_transform_vector3` could be used for calculating the current pose or calculate the control commands for movements. 

We could safely remove `UNINTENDED-MOVEMENT` for `motoman/1ec8ca1/1ec8ca1.bug`

gavanderhoorn, 2021-05-05:
We've indeed extrapolated in all but the last bug report.

But we did that based on typical use-cases in which the buggy components / functions are used.

TF is used to track the position of objects in space, and based on that, in almost all cases robot software will decide "where to go" or "what to do" next. It's not too much of a stretch to see that having consequences such as `UNINTENDED-MOVEMENT`.

Same for (depth)image input: if that's flipped, the robot may (will?) go in the wrong direction.

Incorrect robot model will have two main consequences: wrong display and incorrect FK/IK (ie: positioning) of the robot. The latter will certainly result in incorrect motion if there is any motion (not sure whether we consider that *unintended* though).

@git-afsantos wrote:
> The others are likely to be involved somehow in movement commands [..]

yes, see above.

> but the bug descriptions do not address that as an observed failure.

Correct.

@ipa-hsd wrote:
> We could safely remove `UNINTENDED-MOVEMENT` for `motoman/1ec8ca1/1ec8ca1.bug`

robot support packages are almost exclusively used for planning motions. Incorrect models would lead to incorrect motion.

I'm not sure whether that doesn't then warrant `UNINTENDED-MOVEMENT`. It's a very short bit of extrapolation/inference from my perspective.

ipa-hsd, 2021-05-05:
Well articulated :)

git-afsantos, 2021-05-05:
Summing up, `UNINTENDED-MOVEMENT` is not a direct observation, but will likely happen too in most cases. It makes sense to me, to use this domain knowledge and keep the labels. As long as we are always able to provide a plausible justification, we should be safe.

closed 2021-05-05 by ipa-hsd