robust-rosin/robust #413: geometry2/e12e723: BDO?

gavanderhoorn, 2021-05-12:
I'm unsure as to the classification of this bug.

From the description:

https://github.com/robust-rosin/robust/blob/7d95fc1dcf0eaa7a1fd7a5f5c087ad927f862e16/geometry2/e12e723/e12e723.bug#L3-L17

notice how the classification states *run-time issue*, while the fault codes are:

https://github.com/robust-rosin/robust/blob/7d95fc1dcf0eaa7a1fd7a5f5c087ad927f862e16/geometry2/e12e723/e12e723.bug#L77-L82

Furthermore: from the PR description it's unclear to me how we've concluded the PR solved a build issue: the PR author does add a `build_depend` on `python_orocos_kdl`, but I'm not sure whether that's actually needed. Afaict, `python_orocos_kdl` is only used by Python at runtime (in `tf2_geometry_msgs.py`).

gavanderhoorn, 2021-05-12:
What confuses me is this part from the `description` field:

> tf2_geometry_msgs misses to declare build-time and run-time dependencies on python interface to orocos_kdl library.  This may cause a build  error for the package if built on a system where these libraries are not installed [..]

afaict, `PyKDL` is used at runtime ([here](https://github.com/ros/geometry2/blob/c73b5939723db078c9bbe18523230ad54f859682/tf2_geometry_msgs/src/tf2_geometry_msgs/tf2_geometry_msgs.py#L31)).

This part seems to agree with that:

> Runtime crashes will occur  in python client code that tries to use geometry messages.

ipa-hsd, 2021-05-12:
Well this is embarrassing, but I thought the build error was something to do with the python wrapper library (https://github.com/foolyc/PyKDL/tree/master/python_orocos_kdl) not having installed.

gavanderhoorn, 2021-05-12:
Nothing to be embarrassed about I believe. It's all very terse and sometimes murky descriptions from 5 years ago.

>  but I thought the build error [..]

yes, so *which* build error?

I don't see one mentioned in the PR.

gavanderhoorn, 2021-05-12:
If you look at [CMakeLists.txt](https://github.com/ros/geometry2/blob/3c51d5936805512888bd8ee2893f1de22b287829/tf2_geometry_msgs/CMakeLists.txt) from `indigo-devel`, `python_orocos_kdl` is not mentioned. `orocos_kdl` is, but that's the C++ side.

gavanderhoorn, 2021-05-12:
ah, *maybe* it is a `build_depend` because of [this Python test](https://github.com/ros/geometry2/blob/ba00258027dbfb7cc86e3becd3e457716262d163/tf2_geometry_msgs/CMakeLists.txt#L39), which is run at build time (if tests are enabled).

@ipa-hsd @ChrisTimperley @wasowski @git-afsantos: provided that's correct, would that make this a build error? The dependency isn't actually involved in the build. The *execution* of a test case which happens to use Python fails (probably, still have to confirm), which seems like it's something else than a typical BDO fault.

But maybe my definition of *build failure* is too narrow.

ipa-hsd, 2021-05-12:
Shouldn't it be `<test_depend>` in that case? In either case, I had not noticed the test in `CMakeLists.txt`

gavanderhoorn, 2021-05-12:
The buildfarm is a bit weird with `test_depend`. It could be `build_depend` is needed here.

I also believe the situation around `test` and `build_depend` is different from how it was in 2016.

gavanderhoorn, 2021-05-12:
Decision: remove `BDO:BUILD`, remove `SYSTEM:LOSS-OF-CONTROL`.

Could potentially add something else: @git-afsantos, seeing as you opened #406?

git-afsantos, 2021-05-12:
Your assessment makes sense to me, so :+1: 

I think that I will maintain my opinion from #406; `geometry2` is not a system per se, so, if anything, it might result in `SYSTEM:LOSS-OF-FUNCTIONALITY` (i.e., node tries to use geometry messages, crashes and that part of the system becomes unavailable). But I would also be ok with `SYSTEM:NONE`, if nothing has been observed.

gavanderhoorn, 2021-05-19:
> it might result in `SYSTEM:LOSS-OF-FUNCTIONALITY` (i.e., node tries to use geometry messages, crashes and that part of the system becomes unavailable)

yes, if we extrapolate, this is what *could* happen.

> But I would also be ok with `SYSTEM:NONE`, if nothing has been observed.

The bug report itself does not mention anything related to the runtime consequences of the bug, and we decided to be a bit more conservative: instead of speculating, restrict bug descriptions to whatever is actually described in the original report.

So I'm going to add `SYSTEM:NONE` for now.

gavanderhoorn, 2021-05-19:
Thanks everyone :+1:

commit 29f3ed23 (2021-05-19): e12e723: update fault and failure codes (fix #413)

See the discussion in #413 for context.
closed 2021-05-19 by gavanderhoorn