robust-rosin/robust #414: motoman/377d7be: is this LIVENESS?

gavanderhoorn, 2021-05-19:
[motoman/377d7be](https://github.com/robust-rosin/robust/blob/7e3932c4906a2851d1f84bfd96ecaa58924c9f57/motoman/377d7be/377d7be.bug) currently has a `SOFTWARE:LIVENESS` failure label.

Description:

https://github.com/robust-rosin/robust/blob/7e3932c4906a2851d1f84bfd96ecaa58924c9f57/motoman/377d7be/377d7be.bug#L5-L18

I would agree with that `failure-code` if the system ever reached a state in which it could be started, but the problem reported in the issue occurs during installation. It's impossible to load a specific part of the driver software onto the Yaskawa controller, which makes it impossible to ever start the driver.

Should this be some variant of `BDO` instead?

gavanderhoorn, 2021-05-26:
Related: #416.

ipa-hsd, 2021-06-30:
If the system never was initialized properly, it would be `SOFTWARE:NONE` and `SYSTEM:NONE` I suppose.

ChrisTimperley, 2021-06-30:
SOFTWARE:BDO and SYSTEM:NONE seem appropriate to me

ipa-hsd, 2021-06-30:
Yes `BDO`

gavanderhoorn, 2021-06-30:
It's similar to some other bugs: the system never "got control", so there doesn't appear to be any opportunity to *lose* it.

It's a `BDO` indeed (deployment in this case).

gavanderhoorn, 2021-06-30:
Corrected in 153b315.

commit 153b3153 (2021-06-30): 377d7be: it's an installation issue. Fix #414
closed 2021-06-30 by gavanderhoorn