robust-rosin/robust #427: not a bug: dab1b8a?

gavanderhoorn, 2021-06-30:
Similar to #426. This bug description ([dab1b8a](https://github.com/robust-rosin/robust/blob/93afbcb2ed1e74a9fbfa8eef7c2293ec8526f8d6/mavros/dab1b8a/dab1b8a.bug)) references another commit in the same PR as in #426.

Same question.

ChrisTimperley, 2021-07-07:
Thanks for pointing this out. This is a more general issue: Should we consider bug fixes that occur on PR branches within our dataset? The only repo that should be affected by this issue is MAVROS, since that's the only repo where we looked at commits rather than just PR/issues. Let's figure out how many bugs within our dataset occur on PR branches, and then we can discuss and make a decision on the issue at our next meeting.

gavanderhoorn, 2021-07-07:
There's a really on-topic Github help article about this: [Searching issues and pull requests](https://docs.github.com/en/github/searching-for-information-on-github/searching-on-github/searching-issues-and-pull-requests#search-by-the-commit-shas-within-a-pull-request).

Example: `https://api.github.com/search/issues?q=<INSERT_HASH_HERE>`.

That should result in (a list of) `https://api.github.com/repos/<ORG>/<REPO>/issues/<PR_ID>`.