robust-rosin/robust #446: @gavanderhoorn we're unsure whether this leads to either: (a) a crash at run-time, or (b) using a meaningless/corrupted transform. If the former, then we should go with LOSS-OF-FUNCTIONALITY. If the latter, we should go with BEHAVING-INCORRECTLY.

ChrisTimperley, 2021-09-09:
@gavanderhoorn we're unsure whether this leads to either: (a) a crash at run-time, or (b) using a meaningless/corrupted transform. If the former, then we should go with LOSS-OF-FUNCTIONALITY. If the latter, we should go with BEHAVING-INCORRECTLY.

_Originally posted by @ChrisTimperley in https://github.com/robust-rosin/robust/pull/430#discussion_r678371955_

ChrisTimperley, 2021-09-09:
@gavanderhoorn @ipa-hsd I've moved this issue to here.

gavanderhoorn, 2021-09-14:
If I understand `1b5fa94` correctly, it would lead to a memory leak.

So (at a high-level): if any *Python* node which uses TF2 runs for a sufficient amount of time, it will run out of memory and crash.

From the [linked PR](https://github.com/ros/geometry2/pull/251):

> I have found a memory leak in tf2_py.cpp
> [..]
> Whit this fix the memory leak is fixed.
> We have experience this memory leak when having robots working during days.

gavanderhoorn, 2021-09-14:
Btw: it looks like https://github.com/ros/geometry2/issues/258 is the issue which reported this (although it was reported by a different user).

Perhaps we should update the `.bug`?